project-chip/connectedhomeip #4204: Simplify the management of the "waiting for response" state in ExchangeContext

yufengwangca, 2020-12-15:
#### Problem
        CancelResponseTimer();

        // If the context was expecting a response to a previously sent message, this message
        // is implicitly that response.
        SetResponseExpected(false);

Check if we can simplify the management of the "waiting for response" state in ExchangeContext

 #### Proposed Solution
Simplify the management of the "waiting for response" state in ExchangeContext

bzbarsky-apple, 2020-12-15:
In particular, can we avoid needing two function calls that have to come together to manage the state properly?

yufengwangca, 2021-02-08:
We are cleaning up the pending issue now, I think setting the response expected to false separately after receiving the response make the code easier to understand. 

        // Since we got the response, cancel the response timer.
        CancelResponseTimer();

        // If the context was expecting a response to a previously sent message, this message
        // is implicitly that response.
        SetResponseExpected(false);

Close for now and will reopen later if needed.

closed 2021-02-08 by yufengwangca