project-chip/connectedhomeip #4224: Generate the Rotating Device Id based on Reset Counter

hnnajh, 2020-12-15:
Currently the Rotating Device Id is hard-coded however it should be generated from the device serial number + reset counter

bzbarsky-apple, 2020-12-17:
Note that fixing this is a must-have for shipping rotating device ids.

tima-q, 2021-03-01:
#4362 triggered a runtime issue inside our NVM implementation 

The R/W to a LifetimePersistedCounter that is stored in NVM/Configuration is incorrect.
The key ID for that counter is not known at ConfigurationManager level.
Our NVM/Configuration manager implementation asserts on this unknown key as we only register the expected list.
Other vendor's might have the same issue.

Was there an idea to setup the relation between the new lifetimecounter and the configuration manager defines ?
There is currently no direct link between the define for the key (CHIP_CONFIG_LIFETIIME_PERSISTED_COUNTER_KEY)
and the list of key ID's listed in the different src\platform\xyz\xyzConfig.h files

We will disable the feature for now at build-time to proceed, but I think the structural solution would be to introduce a specific key in the ConfigurationManager for this counter. 
@hnnajh What are/were your thoughts around this in terms of implementation ?

bzbarsky-apple, 2022-05-04:
@hnnajh What remains to be done here?

chrisdecenzo, 2022-05-16:
It looks like rotating ID unique id input is coming from a #define (CHIP_DEVICE_CONFIG_ROTATING_DEVICE_ID_UNIQUE_ID). This needs to be something that device makers can populate at run time from storage.

PR #18514 (2022-05-18): Address Rotating ID issues
#### Problem
Currently, the rotating id is generated from a #define unique id
* Fixes #4224 

#### Change overview
Add platform command to set the rotating id unique id so that platforms can call this at startup

#### Testing
* Tested using tv-app and tv-casting-app
closed 2022-05-18 by andy31415