project-chip/connectedhomeip #4225: Chip Tool unable to complete BLE pairing for nordic dev board running Lighting App on Mac

SaileeG, 2020-12-16:
Setup:
- nRF52840 DK board running Lighting Sample App
- Made sure the BLE Advertisements are enabled on the Board after Flashing.
- Able to see the the nRF52840 DK Board with Lighting App as ChipLight using my Phone and Mac

When using the CHIP tool to pair over BLE I see LED1 Rapid Even Flashing indicating The device is in the unprovisioned state and a commissioning application is connected through Bluetooth LE. The console logs show 'Secure Pairing Success' and then loses connection to the board.

`skorgaonkar-macbookpro2:debug skorgaonkar$ ./chip-tool pairing ble SSID_XXX PASSWORD_YYY 12345678 3840
CHIP: [DL] _Init
CHIP: [IN] TransportMgr initialized
CHIP: [IN] local node id is 112233

CHIP: [BLE] NewConnection
CHIP: [DL] CHIP task running
CHIP: [BLE] CBManagerState: ON
CHIP: [BLE] Connecting to device with discriminator: 3840
CHIP: [BLE] subscribe complete, ep = 0x10ab03b98
CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
CHIP: [BLE] using BTP fragment sizes rx 20 / tx 20.
CHIP: [BLE] local and remote recv window size = 3
CHIP: [BLE] Sent PBKDF param request
CHIP: [BLE] Received rendezvous message from 12344321
CHIP: [BLE] Received PBKDF param response
CHIP: [BLE] Sent spake2p msg1
CHIP: [BLE] Received rendezvous message from 12344321
CHIP: [BLE] Received spake2p msg2
CHIP: [BLE] Sent spake2p msg3
CHIP: [CTL] Remote device completed SPAKE2+ handshake

CHIP: [TOO] OnNetworkCredentialsRequested
CHIP: [NP] Sending Network Creds. Delegate 0x7fada6009018

CHIP: [TOO] Secure Pairing Success
CHIP: [TOO] No response from device
CHIP: [CTL] Shutting down the commissioner
CHIP: [CTL] Shutting down the controller`

cjandhyala, 2020-12-16:
Mac side logs

uart:~$ D: 28113[DL] SetAdvertisingEnabled(true)                             
I: Enabled BLE Advertisement                                                 
D: 28119[DL] In DriveBLEState                                
I: 28122[DL] Setup PIN discriminator not found; using default: f00
E: 28128[DL] void chip::DeviceLayer::Internal::GenericThreadStackManagerImpl_Zephyr<ImplClass>::_OnCHIPoBLEAdvertisingStart() [with ImplClass = chip::DeviceLayer::ThreadStackManagerImpl]: NOT IMPLEMENTED
I: 28147[DL] CHIPoBLE advertising started                    
I: 42851[DL] BLE connection established (ConnId: 0x00)       
I: 42856[DL] Current number of connections: 1/1              
D: 42860[DL] In DriveBLEState
I: 42863[DL] CHIPoBLE advertising stopped
E: 42867[DL] void chip::DeviceLayer::Internal::GenericThreadStackManagerImpl_Zephyr<ImplClass>::_OnCHIPoBLEAdvertisingStop() [with ImplClass = chip::DeviceLayer::ThreadStackManagerImpl]: NOT IMPLEMENTED
D: 43238[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
I: 43245[BLE] local and remote recv window sizes = 3
I: 43250[BLE] selected BTP version 3
I: 43253[BLE] using BTP fragment sizes rx 20 / tx 20.
D: 43328[DL] ConnId: 0x00, New CCCD value: 0x0002
D: 43333[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 6)
I: 43340[DL] CHIPoBLE connection established (ConnId: 0x00, GATT MTU: 23)
D: 43373[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 43381[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 43403[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 43410[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
D: 43448[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 43456[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 43478[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 43485[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
I: 43493[BLE] Received rendezvous message from lu
D: 43497[BLE] Received PBKDF param request
D: 43726[BLE] Sent PBKDF param response
E: 43729[DL] Long dispatch time: 251 ms
D: 43763[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 43771[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 43808[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 43817[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 43824[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 43853[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 43861[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 43898[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 43907[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 43914[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 43943[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 43951[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 19)
D: 43988[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 43997[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 44033[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 44040[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
D: 44078[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 44086[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 44108[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 44115[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
D: 44153[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 44161[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 44183[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 44190[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
D: 44228[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 44237[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
I: 44244[BLE] Received rendezvous message from lu
D: 44248[BLE] Received spake2p msg1
D: 44405[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 44413[BLE] Sent spake2p msg2
E: 44416[DL] Long dispatch time: 179 ms
D: 44438[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 44447[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 44454[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 44483[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 44491[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 44528[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 44537[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
[... 47 lines cut ...]
D: 55255[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 57742[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 60249[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
[... cut ...]
D: 85675[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 88162[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 90669[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
D: 90745[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 93232[DL] Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 95740[DL] Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
D: 95815[DL] Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)

kkasperczyk-no, 2020-12-17:
The problem is probably related to the lack of Thread provisioning implementation on Linux CHIP Tool version, as on the iOS similar problem was observed. Currently only Android platform is supporting Thread provisioning, while other assumes that Wi-Fi provisioning is the only one existing.
This issue is someway related do this one: #3830 
@erjiaqing as far as I know, you were going to work on adding Thread provisioning, but do you also consider doing it on Linux platform? If not maybe you know about any plans related to this?

erjiaqing, 2020-12-18:
> The problem is probably related to the lack of Thread provisioning implementation on Linux CHIP Tool version, as on the iOS similar problem was observed. In that case part of Rendezvous is working properly, but then device is not able to get Thread network credentials from the CHIP Tool app. Currently only Android platform is supporting Thread provisioning, while other assumes that Wi-Fi provisioning is the only one existing.
> This issue is someway related to this one: #3830
> @erjiaqing as far as I know, you were going to work on adding Thread provisioning, but do you also consider doing it on Linux platform? If not maybe you know about any plans related to this?

In my plan, the first pr is to enable basic network provisioning (wifi / thread) on linux platform ChipController/chip-device-ctrl, other platforms should be a follow up.

One thing that blocking is we need to modify BLE transport so ZCL commands can go through it.

SaileeG, 2021-01-15:
> The problem is probably related to the lack of Thread provisioning implementation on Linux CHIP Tool version, as on the iOS similar problem was observed. In that case part of Rendezvous is working [...]

Is Thread provisioning implementation on Linux not supported for Python Controller as well?

franck-apple, 2021-04-06:
CSG not using CHIP tool anymore. Closing

closed 2021-04-06 by franck-apple