project-chip/connectedhomeip #4226: Chip Tool unable to complete BLE pairing for nordic dev board running Lighting App with Pigweed RPCs on Mac

SaileeG, 2020-12-16:
Setup:
- nRF52840 DK board flashed with Lighting Sample App with Pigweed RPC(https://github.com/project-chip/connectedhomeip/pull/3930)
- Enabled BLE Advertisements by pressing Button 4 on  nRF52840 DK(Done over Pigweed RPC)
- Made sure the BLE Advertisements are enabled on the Board after Flashing.
- Able to see the the nRF52840 DK Board with Lighting App with RPC as ChipLight using my Phone and Mac

Unable to pair Lighting App with Pigweed RPCs on Nordic dev board over BLE.

skorgaonkar-macbookpro2:debug skorgaonkar$ ./chip-tool pairing ble SSID_XXX PASSWORD_YYY 12345678 3840
CHIP: [DL] _Init
CHIP: [IN] TransportMgr initialized
CHIP: [IN] local node id is 112233

CHIP: [BLE] NewConnection
CHIP: [DL] CHIP task running
CHIP: [BLE] CBManagerState: ON
CHIP: [BLE] Connecting to device with discriminator: 3840
CHIP: [BLE] subscribe complete, ep = 0x106aa0b98
CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
CHIP: [BLE] using BTP fragment sizes rx 20 / tx 20.
CHIP: [BLE] local and remote recv window size = 3
CHIP: [BLE] Sent PBKDF param request
CHIP: [TOO] No response from device
CHIP: [CTL] Shutting down the commissioner
CHIP: [CTL] Shutting down the controller

SaileeG, 2021-01-08:
Retried with the latest master and the process is getting further and failing at a later stage similar to the failure observed on MAC tracked in https://github.com/project-chip/connectedhomeip/issues/4225

skorgaonkar-macbookpro2:debug skorgaonkar$ ./chip-tool pairing ble SSID_XXX PASSWORD_YYY 12345678 3840
CHIP: [DL] _Init
CHIP: [IN] TransportMgr initialized
CHIP: [IN] local node id is 112233

CHIP: [DL] CHIP task running
CHIP: [BLE] NewConnection
CHIP: [BLE] CBManagerState: ON
CHIP: [BLE] Connecting to device with discriminator: 3840
CHIP: [BLE] subscribe complete, ep = 0x10d4efb98
CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
CHIP: [BLE] using BTP fragment sizes rx 20 / tx 20.
CHIP: [BLE] local and remote recv window size = 3
CHIP: [BLE] Sent PBKDF param request
CHIP: [BLE] Received PBKDF param response
CHIP: [BLE] Sent spake2p msg1
CHIP: [BLE] Received spake2p msg2
CHIP: [BLE] Sent spake2p msg3
CHIP: [CTL] Remote device completed SPAKE2+ handshake

CHIP: [TOO] OnNetworkCredentialsRequested
CHIP: [NP] Sending Network Creds. Delegate 0x7fe5ab25e818

CHIP: [TOO] Secure Pairing Success
CHIP: [TOO] No response from device
CHIP: [CTL] Shutting down the commissioner
CHIP: [CTL] Shutting down the controller

Damian-Nordic, 2021-01-08:
Hi,
The reason is that the Nordic dev board is a Thread device, so it can't handle WiFi credentials sent during the pairing procedure. 

What needs to be implemented to enable provisioning Thread devices is another command, for example:
```
pairing ble-thread THREAD-CHANNEL THREAD-PANID THREAD-XPANID THREAD-MASTERKEY 12345678 3840
```
which would work just like `pairing-ble`, but once pairing gets to the provisioning phase, it would send Thread settings instead of Wi-Fi credentials.

LuDuda, 2021-03-04:
Are we considering using and extending the `chip-tool` anymore? The legacy Thread provisioning (same as in WiFi) has been recently added to the Python CHIP Controller.

RoyAnupam, 2021-09-25:
> Hi,
> The reason is that the Nordic dev board is a Thread device, so it can't handle WiFi credentials sent during the pairing procedure.
>
> What needs to be implemented to enable provisioning Thread [...]

@Damian-Nordic , @LuDuda 
 I am currently testing the thread commissioning over BLE feature. Facing some issue in parsing OperationalDataset in the chip-lighing-app.
D/CHIP    ( 1850): ZCL: Received Cluster Command: Cluster=31 Command=6 Endpoint=0
D/CHIP    ( 1850): ZCL: Failed to parse Thread operational dataset: CHIP Error 4047 (0x00000FCF): Invalid argument
D/CHIP    ( 1850): ZCL: AddThreadNetwork: 19

Following is the Thread Network & corresponding chip-tool command, that I used.

**network info:**
> scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
> | 0 | OpenThread-09a9  | 2b44e353c40c2432 | 09a9 | 16d1ef880e8ed012 | 21 | -56 | 156 |
Done

**chip-tool command**
./chip-tool pairing ble-thread  0x1509a92b44e353c40c243212345678 0 12341234 291

Basically appended all bytes one after another in the operational dataset argument, but I am sure, the byte span passed  is messed up. Could you share some light about correct way to form this argument with above Network info as example? Thank You very much!

closed 2022-01-25 by woody-apple