project-chip/connectedhomeip #4228: Chip Tool unable to complete BLE pairing for Nordic dev board running Lighting App on Raspi

SaileeG, 2020-12-16:
Setup:
- nRF52840 DK board running Lighting Sample App
- Made sure the BLE Advertisements are enabled on the Board after Flashing.
- Able to see the the nRF52840 DK Board with Lighting App as ChipLight on Raspi Bluetooth settings

ubuntu@ubuntu:~/CHIP/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool pairing ble SSID_XXX PASSWORD_YYY 12345678 3840
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

CHIP:DL: Avahi client registered
CHIP:DL: CHIP task running
CHIP:DL: FAIL: BLE name is NULL in ConfigureBluezAdv
CHIP:DL: Disabling CHIPoBLE service due to error: Error 4003 (0x00000FA3)
CHIP:TOO: No response from device
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller

SaileeG, 2021-01-11:
As suggested by Zoltan removed .environment and reran 'source scripts/bootstrap.sh '
Still see the same issue.

ubuntu@ubuntu:~/chip1/connectedhomeip$ source scripts/bootstrap.sh 

  WELCOME TO...

  ▄███▒  ░▓█  ░▓█ ░▓█▓ ▒█████▄
 ██▒ ▀█▒  ▒█   ▒█  ░█▒  ▒█░  █░
 █▓░      ▒██████  ░█▒  ▒█▄▄▄█░
 ▓█   █▒  ▒█   ▒█  ░█░  ▒█▀
 ░▓███▀  ░▓███░▓█▒ ░█░  ▒█

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done
  Setting up Python environment.....[|]
ninja: Entering directory `/home/ubuntu/chip1/connectedhomeip/.environment/pigweed-venv/repo-3e76f0bc75f3bf878e929104ca465277'
[1/48] ACTION //pw_bloat/py:py(//pw_toolchain/dummy:dummy)
[2/48] ACTION //pw_arduino_build/py:py(//pw_toolchain/dummy:dummy)
[3/48] ACTION //pw_cli/py:py(//pw_toolchain/dummy:dummy)
[4/48] ACTION //pw_build/py:py(//pw_toolchain/dummy:dummy)
[5/48] ACTION //pw_docgen/py:py(//pw_toolchain/dummy:dummy)
[6/48] ACTION //pw_cpu_exception_cortex_m/py:py(//pw_toolchain/dummy:dummy)
[7/48] ACTION //pw_doctor/py:py(//pw_toolchain/dummy:dummy)
[8/48] ACTION //pw_env_setup/py:py(//pw_toolchain/dummy:dummy)
[9/48] ACTION //pw_module/py:py(//pw_toolchain/dummy:dummy)
[10/48] ACTION //pw_allocator/py:py(//pw_toolchain/dummy:dummy)
[11/48] ACTION //pw_package/py:py(//pw_toolchain/dummy:dummy)
[12/48] ACTION //pw_status/py:py(//pw_toolchain/dummy:dummy)
[13/48] ACTION //pw_protobuf/py:py(//pw_toolchain/dummy:dummy)
[14/48] ACTION //pw_protobuf_compiler/py:py(//pw_toolchain/dummy:dummy)
[15/48] ACTION //pw_tokenizer/py:py(//pw_toolchain/dummy:dummy)
[16/48] ACTION //pw_presubmit/py:py(//pw_toolchain/dummy:dummy)
[17/48] ACTION //pw_trace_tokenized/py:py(//pw_toolchain/dummy:dummy)
[18/48] ACTION //pw_trace/py:py(//pw_toolchain/dummy:dummy)
[19/48] ACTION //pw_rpc/py:py(//pw_toolchain/dummy:dummy)
[20/48] ACTION //pw_watch/py:py(//pw_toolchain/dummy:dummy)
[21/48] ACTION //pw_unit_test:unit_test_proto.python._gen(//pw_toolchain/dummy:dummy)
FAILED: gen/pw_unit_test/protos/pw_unit_test_proto/unit_test_pb2.py 
python ../../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root /home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/ --current-path /home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_unit_test --default-toolchain=//pw_toolchain/dummy:dummy --current-toolchain=//pw_toolchain/dummy:dummy --capture-output -- /home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_protobuf_compiler/py/pw_protobuf_compiler/generate_protos.py --language python --module-path /home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_unit_test --include-file /home/ubuntu/chip1/connectedhomeip/.environment/pigweed-venv/repo-3e76f0bc75f3bf878e929104ca465277/gen/pw_unit_test/unit_test_proto_includes.txt --out-dir /home/ubuntu/chip1/connectedhomeip/.environment/pigweed-venv/repo-3e76f0bc75f3bf878e929104ca465277/gen/pw_unit_test/protos /home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_unit_test/pw_unit_test_proto/unit_test.proto
Traceback (most recent call last):
  File "/home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_protobuf_compiler/py/pw_protobuf_compiler/generate_protos.py", line 190, in <module>
    sys.exit(main())
  File "/home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_protobuf_compiler/py/pw_protobuf_compiler/generate_protos.py", line 166, in main
    process = subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'protoc'
[22/48] ACTION //pw_hdlc_lite/py:py(//pw_toolchain/dummy:dummy)
[23/48] ACTION //pw_bloat/py:py.install(//pw_toolchain/dummy:dummy)
ninja: build stopped: subcommand failed.

Traceback (most recent call last):
  File "/home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 593, in <module>
    sys.exit(main())
  File "/home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 585, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 307, in setup
    result = step()
  File "/home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 417, in virtualenv
    env=self._env,
  File "/home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 244, in install
    install_packages(gn_target)
  File "/home/ubuntu/chip1/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 229, in install_packages
    ins.read())
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/ubuntu/chip1/connectedhomeip/.environment/pigweed-venv/repo-3e76f0bc75f3bf878e929104ca465277', ':python.install']' returned non-zero exit status 1
Error during bootstrap--see messages above.
ubuntu@ubuntu:~/chip1/connectedhomeip$

Shuyang-z, 2021-02-04:
Same error on Raspberry 3B running Ubuntu server 20.04.1 LTS

Damian-Nordic, 2021-03-04:
@SaileeG The error you mentioned in the description was generated because at the time BLE central implementation was missing on Linux platform. Now the error should be gone. Thread provisioning is still unimplemented in `chip-tool`, because I was under impression that the whole community is moving towards CHIP python controller. Let me know if you still need Thread provisioning in `chip-tool` and if the original issue still occurs.

nageshshamnur, 2021-04-28:
I am facing similar issue running tv-app. 

Branch: Test_event_2.2

when i build and run tv-app under examples/tv-app/linux on Raspberrypi4B with native ble chipset present on the board (hci0) with the following command:
```
connectedhomeipexamples/tv-app/linux$ sudo out/debug/chip-tv-app --ble-device 0
```
error as below:
```
......
CHIP:DL: FAIL: BLE name is NULL in ConfigureBluezAdv
CHIP:DL: Disabling CHIPoBLE service due to error: CHIP Error 4003 (0x0000FA3): Incorrect state
```

closed 2022-01-25 by woody-apple

gautam1joshi, 2023-06-12:
is anyone have a solution of this? i also have the same issue.