project-chip/connectedhomeip #4235: Chip-tool to support commissioning of Nodes already on the customer’s IP network

cjandhyala, 2020-12-16:
Chip-tool to support commissioning of Nodes already on the customer’s IP network (such as Ethernet-connected Nodes, or Wi‐Fi Nodes already associated to the Wi‐Fi network via other means) Using UDP/IP for CHIP

mspang, 2021-02-25:
@erjiaqing How much of this is done from https://github.com/project-chip/connectedhomeip/pull/4050 ?

erjiaqing, 2021-02-25:
> @erjiaqing How much of this is done from #4050 ?

As I have talked about this issue before, the commissioning over IP will be implemented as described in the spec (aka using clusters).

This was not done before since we don't have a solution to make both IM command and current ember library work, I just settled the plan for resolving this with @vivien-apple , as described in https://github.com/project-chip/connectedhomeip/issues/5011, I will start working on this as soon as I made some progress that PR.

erjiaqing, 2021-02-26:
Let's use NetworkCommissioningCluster when using IP, this matches the spec. :)

erjiaqing, 2021-03-03:
Close this issue since we can use network commissioning cluster now, thread support will be added soon.

closed 2021-03-03 by erjiaqing