project-chip/connectedhomeip #4236: Chip tool to support Run-time dynamic discovery of operational CHIP Nodes

cjandhyala, 2020-12-17:
A CHIP controller should discover the current IPv6 address and port for a known commissioned CHIP Node.

Damian-Nordic, 2021-03-31:
@cjandhyala Note that Python CHIP controller built with `--args='chip_mdns="platform"'` supports "resolve" command which does more less what the description contains. Is there anything more needed for this task?

cjandhyala, 2021-04-14:
@Damian-Nordic , building with the above args ,crashes the python controller

$ chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
Traceback (most recent call last):
  File "/home/ubuntu/.local/bin/chip-device-ctrl", line 642, in <module>
    main()
  File "/home/ubuntu/.local/bin/chip-device-ctrl", line 623, in main
    devMgrCmd = DeviceMgrCmd(rendezvousAddr=options.rendezvousAddr, controllerNodeId=options.controllerNodeId, bluetoothAdapter=adapterId)
  File "/home/ubuntu/.local/bin/chip-device-ctrl", line 146, in __init__
    self.devCtrl = ChipDeviceCtrl.ChipDeviceController(controllerNodeId=controllerNodeId, bluetoothAdapter=bluetoothAdapter)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/chip/ChipDeviceCtrl.py", line 51, in wrapper
    instance[0] = cls(*args, **kwargs)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/chip/ChipDeviceCtrl.py", line 84, in __init__
    raise self._ChipStack.ErrorToException(res)
chip.exceptions.ChipStackError: CHIP Error 4003 (0x00000FA3): Incorrect state

Damian-Nordic, 2021-04-15:
@cjandhyala It should be fixed by https://github.com/project-chip/connectedhomeip/pull/6041.

cjandhyala, 2021-04-17:
@Damian-Nordic , looks like the resolve command is looking for mdns _chip_tcp only. Could we expand this to cover _chipd._udp. so that we can verify '4.2.3. Commissionable Discovery' as well

Damian-Nordic, 2021-04-19:
@cjandhyala That is a separate functionality and I believe @cecille is working on adding another command for that purpose: #5716. This task is only about operational discovery which uses `_chip._tcp` service type.

franck-apple, 2021-05-17:
@Damian-Nordic, was this solved with #6459 (that replaced #5716)?

closed 2021-07-21 by Damian-Nordic