project-chip/connectedhomeip #4257: Update Build.gn files to include modules from the top level src

hnnajh, 2020-12-18:
Build.gn files include header files / dependencies by adding their immediate parents instead of referencing them by the top src folders, which might create some name collisions, it's better to include the top folder so when including the headers we can reference the full path under src folder

Example of the current state:
# This is a real toolchain. Build CHIP.
  group("default") {
    deps = [
      "${chip_root}/src/app",
      "${chip_root}/src/ble",
      "${chip_root}/src/controller",
      "${chip_root}/src/crypto",
      "${chip_root}/src/inet",
      "${chip_root}/src/lib",
      "${chip_root}/src/lib/asn1",
      "${chip_root}/src/lib/core",
      "${chip_root}/src/lib/shell",
      "${chip_root}/src/lib/support",
      "${chip_root}/src/lwip:all",
      "${chip_root}/src/messaging",
      "${chip_root}/src/protocols",
      "${chip_root}/src/setup_payload",
      "${chip_root}/src/system",
      "${chip_root}/src/transport",
      "${mbedtls_root}:mbedtls",
      "${nlassert_root}:nlassert",
      "${nlio_root}:nlio",
      "${nlunit_test_root}:nlunit-test",
    ]

if any of those two folder have header with the same name, a name collision will happen.

bzbarsky-apple, 2020-12-18:
@mspang do deps get automatically added to `-I`?  I thought that was what `include_dirs` was for....

mspang, 2020-12-18:
No, nothing is automatically added to the include path, there are all added via include_dirs.

automake had almost every directory as an include dir and those were all grandfathered in the GN migration. I think this bug amounts to removing those.

Additionally, I believe we should really move everything to src/chip/ or just rename "src" to "chip",  so that we avoid similar conflicts with other projects:

`#include <chip/system/SystemLayer.h>
`

mspang, 2020-12-18:
Actually I fixed a number of these preliminary to GN migration, but apparently not all.

bzbarsky-apple, 2020-12-18:
@hnnajh Which directory were you in when you ran into over-broad `include_dirs`?

mspang, 2020-12-18:
Wasn't it setup_payload ?

Just need to remove these lines:

https://github.com/project-chip/connectedhomeip/blob/0e55cf87551ec632592221e2969001c98228796d/src/setup_payload/BUILD.gn#L17-L19

woody-apple, 2021-02-02:
@hnnajh what are the next steps here?