project-chip/connectedhomeip #4270: Browser support

backkem, 2021-01-03:
#### Problem
The IoT space and the browser world have historically been heavily divided. Since this project aims for a new standard, running over IP, it may be a good time to consider bridging these two worlds and finally making IoT devices 'approachable' from the browser. The goal is to enable websites and PWAs to interact with IoT directly, over the local LAN, without relying on the cloud.

With current browser technology this is not possible. It is trivial to talk to a remote server in the cloud but there is no user-friendly way of talking to a TV, NAS or thermostat sitting right next to you. This could be different.

 #### Proposed Solution
 ### Cloud
To talk to a device that's sitting right next to you there should be no reliance on a remote server. We should design our systems to be more resilient.

 ### User experience
To enable broad adoption, it should be trivial to connect to a device on LAN. However, this is currently not possible. One could use HTTP however, the browser will mark the page as insecure with ever increasingly scary icons. In addition, unencrypted traffic should be considered a bad practice these days, even on LAN. The next option would be self-signed certificates. However, this quickly becomes overly complex for the average user. You have to either accept a scary 'insecure website' prompt or add the certificate to the certificate store. Neither are acceptable for a regular user.
Instead we should be creating an experience similar to what ChromeCast does; guiding the user to safely connect with a device on LAN.

 ### Thrust
The first problem that has to be solved is thrust. A browser has to trust a device on the local LAN. Since these devices are private, the browser cannot rely on some certificate authority. Instead, a pairing mechanism could be used. Similar to Bluetooth pairing, the browser could maintain a list of thrusted devices. During the pairing TLS certificates are exchanged. These can subsequently be used to secure follow-up communication. Due to a browsers strict security, this pairing mechanism would have to be provided by the browser itself. It likely cannot be done using application (E.g.: JavaScript) code.

 ### Protocol
Due to a browsers strict security, it is not possible to allow direct access to TCP or UDP. One would have to either use HTTPS or add support for specific application level protocols. I think both have merit. Once you exchanged certificates as detailed above, the browser can consider them as part of the certificate store and mark the page as trusted. In addition, new JavaScript APIs can provide access to the CHIP protocol(s) directly.

 ### Alternatives considered
 ## WebRTC
The closest you can get to this promise today would be to setup a P2P connection between the browser and device using WebRTC Data Channel. However, there are still issues with this setup. First, this still requires a public server for 'signaling'. Secondly, WebRTC is a rather heavy-handed protocol for this use-case.
This proposal could also be used in tandem with WebRTC by enabling the 'signaling' step to occur over LAN.

wieweet, 2021-01-30:
I made a suggestion for a spec here: https://discourse.wicg.io/t/idea-local-devices-api-lan-services/5056/8

It's based on the Open Screen Protocol. Any feedback is highly appreciated. This first spec suggestion doesn't take into account compatibility with Project Connected Home over IP. We need your input if this is at all desired.

backkem, 2021-01-30:
I made a suggestion for a spec here: https://discourse.wicg.io/t/idea-local-devices-api-lan-services/5056/8

It's based on the Open Screen Protocol. Any feedback is highly appreciated. This first spec suggestion doesn't take into account compatibility with Project Connected Home over IP. We need your input if this is at all desired.

backkem, 2022-09-16:
I still think there is merit to bringing Matter to the browser. Marking as stale feels somewhat inappropriate for issues tagged as 'future'.

closed 2023-03-25 by stale[bot]

rektide, 2024-01-09:
Can we reopen? @stale bot is such an awful tyrant.

backkem, 2024-01-09:
There is certainly still interest in this topic. I updated the [OP](https://github.com/project-chip/connectedhomeip/issues/4270#issue-777614817) to reflect the latest thinking. From the browser side, we'd like to explore potential synergies between the Matter and the Open Screen Protocol as discussed in w3c/openscreenprotocol#308.

We're looking for a good way to continue this conversation between the standards efforts/organisations.