project-chip/connectedhomeip #4275: Remove map from IM invoke command

yunhanw-google, 2021-01-06:
Are we really OK to use std::map here? It's big and directly uses the heap, right? Believe we're planning to avoid the STL altogether.

_Originally posted by @mspang in https://github.com/project-chip/connectedhomeip/pull/4241#discussion_r552359944_

erjiaqing, 2021-01-22:
According to conversations with @vivien-apple , I will submit another PR as a follow up, in that PR, I will add DispatchSingleCommand(clusterid, commandId, role, tlvReader, commandObj) to replace InteractionModelEngine::ProcessCommand (and this will remove the std::map in InteractionModelEngine, and modify related tests.

In #4298

woody-apple, 2021-02-02:
@yunhanw-google is this now complete with https://github.com/project-chip/connectedhomeip/pull/4298 ?

yunhanw-google, 2021-02-03:
yes, it is completed with #4585

closed 2021-02-03 by yunhanw-google