project-chip/connectedhomeip #4300: CHIP Tool crashes during BLE connection

doru91, 2021-01-08:
#### Problem
CHIP Tool crashes with a NULL Pointer exception while trying to establish a BLE connection with NXP K32W board.
Looking into Wireshark traces at the BLE packets everything looks correct. I attached the results for testing with 2 different phones:

- Xioami Redmi 9:
[xiaomi_debug_chip_tool_failure.zip](https://github.com/project-chip/connectedhomeip/files/5788192/xiaomi_debug_chip_tool_failure.zip)

- Samsung Galaxy S8
[galaxy_s8_debug_chip_tool_null_pointer.zip](https://github.com/project-chip/connectedhomeip/files/5788193/galaxy_s8_debug_chip_tool_null_pointer.zip)

From my understanding Samsung is using its proprietary BLE stack and that's why I decided to use an Xiaomi phone which is using the generic BLE stack.

The NULL Pointer Exception text is this one:

```
2021-01-08 08:03:17.929 27173-27173/com.google.chip.chiptool D/CTL: beginConnectDevice() called with connection object and pincode
2021-01-08 08:03:17.930 27173-27173/com.google.chip.chiptool D/CTL: Received GetMTU
2021-01-08 08:03:17.930 27173-27173/com.google.chip.chiptool D/CTL: Calling Java onGetMTU
2021-01-08 08:03:17.930 27173-27173/com.google.chip.chiptool D/AndroidChipStack: Android Manufacturer: (samsung)
2021-01-08 08:03:17.930 27173-27173/com.google.chip.chiptool D/AndroidChipStack: Android Model: (SM-G950F)
2021-01-08 08:03:17.930 27173-27173/com.google.chip.chiptool D/CTL: Received SendCharacteristic
2021-01-08 08:03:17.931 27173-27173/com.google.chip.chiptool D/CTL: Calling Java SendCharacteristic
2021-01-08 08:03:17.931 27173-27173/com.google.chip.chiptool E/CTL: Java exception thrown in HandleSendCharacteristic
2021-01-08 08:03:17.932 27173-27173/com.google.chip.chiptool W/System.err: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothGattCharacteristic.setValue(byte[])' on a null object reference
2021-01-08 08:03:17.933 27173-27173/com.google.chip.chiptool W/System.err:     at chip.devicecontroller.AndroidChipStack.onSendCharacteristic(AndroidChipStack.java:262)
2021-01-08 08:03:17.933 27173-27173/com.google.chip.chiptool W/System.err:     at chip.devicecontroller.ChipDeviceController.beginConnectDevice(Native Method)
2021-01-08 08:03:17.934 27173-27173/com.google.chip.chiptool W/System.err:     at chip.devicecontroller.ChipDeviceController.beginConnectDeviceBle(ChipDeviceController.java:61)
2021-01-08 08:03:17.934 27173-27173/com.google.chip.chiptool W/System.err:     at com.google.chip.chiptool.provisioning.DeviceProvisioningFragment$startConnectingToDevice$1.invokeSuspend(DeviceProvisioningFragment.kt:101)
2021-01-08 08:03:17.934 27173-27173/com.google.chip.chiptool W/System.err:     at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
2021-01-08 08:03:17.934 27173-27173/com.google.chip.chiptool W/System.err:     at kotlinx.coroutines.DispatchedTask.run(Dispatched.kt:241)
2021-01-08 08:03:17.935 27173-27173/com.google.chip.chiptool W/System.err:     at android.os.Handler.handleCallback(Handler.java:873)
2021-01-08 08:03:17.935 27173-27173/com.google.chip.chiptool W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:99)
2021-01-08 08:03:17.935 27173-27173/com.google.chip.chiptool W/System.err:     at android.os.Looper.loop(Looper.java:214)
2021-01-08 08:03:17.939 27173-27173/com.google.chip.chiptool W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:7050)
2021-01-08 08:03:17.939 27173-27173/com.google.chip.chiptool W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
2021-01-08 08:03:17.940 27173-27173/com.google.chip.chiptool W/System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
2021-01-08 08:03:17.940 27173-27173/com.google.chip.chiptool W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:965)
2021-01-08 08:03:17.941 27173-27173/com.google.chip.chiptool D/DeviceProvisioningFragment: onPairingComplete: 6007
2021-01-08 08:03:17.941 27173-27173/com.google.chip.chiptool D/BLE: Auto-closing end point's BLE connection.
2021-01-08 08:03:17.941 27173-27173/com.google.chip.chiptool D/CTL: Received CloseConnection
2021-01-08 08:03:17.941 27173-27173/com.google.chip.chiptool D/CTL: Calling Java CloseConnection
2021-01-08 08:03:17.942 27173-27173/com.google.chip.chiptool D/ChipDeviceController: Closing GATT and removing connection for 1
2021-01-08 08:03:17.942 27173-27173/com.google.chip.chiptool D/BluetoothGatt: close()
2021-01-08 08:03:17.949 27173-27173/com.google.chip.chiptool D/BluetoothGatt: unregisterApp() - mClientIf=8
2021-01-08 08:03:17.953 27173-27173/com.google.chip.chiptool D/ChipDeviceController: Calling onCloseBleComplete()
2021-01-08 08:03:17.953 27173-27173/com.google.chip.chiptool D/DeviceProvisioningFragment: onCloseBleComplete
2021-01-08 08:03:17.953 27173-27173/com.google.chip.chiptool W/e.chip.chiptoo: 0xebadde09 skipped times: 0
2021-01-08 08:03:17.955 27173-27173/com.google.chip.chiptool A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x3000052 in tid 27173 (e.chip.chiptool), pid 27173 (e.chip.chiptool)
```

Looking into the Wireshark traces it seems that the BLE traffic stops after the Exchange MTU Request/MTU Response: 
[image]

Service discovery seems to be working ok: inside onServicesDiscovered function I added some debug code and the services/characteristics discovered by CHIP Tool are the one that I declared in my GATT database on the accessory side:

 #### Proposed Solution
1. Could you please confirm that the BLE traffic generated on the K32W side looks ok?
2. Do you have any suggestion on how to fix the issue on the CHIP Tool side.

cc @vidhis88 @tcarmelveilleux @vzhodzishsky @LuDuda

Damian-Nordic, 2021-01-08:
@doru91 I don't think it's an issue on the CHIP Tool side (although the error might be handled a bit more gracefully). It works correctly with Nordic and ESP32 devices and the error suggests that the tool is unable to find CHIP RX/TX characteristics on the peer device. You probably should double-check definition of the BLE service on the K32W side (if UUIDs are like on the other platforms etc.).

doru91, 2021-01-08:
@Damian-Nordic, thanks for the reply. I agree that the tool is working with other vendor implementation but maybe it's the packet sequence in our BLE stack that is triggering a race condition. I checked the Wireshark traffic and it looks ok to me. Could you provide me a Wireshark trace with the BLE traffic for your platform? BR!

doru91, 2021-01-11:
Solved, the endianess of the UUID from GATT DB on my side was incorrect.

vidhis88, 2021-01-12:
Closing this as the issue is solved.

closed 2021-01-12 by vidhis88