project-chip/connectedhomeip #4310: CHIP Certificate Code Mixes ASN1_ERROR and CHIP_ERROR Error Spaces

emargolis, 2021-01-11:
Some functions in the CHIP certificate code in this PR https://github.com/project-chip/connectedhomeip/pull/4060 mix ASN1_ERROR and CHIP_ERROR. 
Those have different value spaces. Need to investigate if it is OK to mix them.

If mixing is OK need to document that in the spec

 

#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2021-02-05:
In particular, we have methods for converting CHIP_ERROR to/from device error codes that assume CHIP_ERROR is in its defined range, afaict.

franck-apple, 2021-02-05:
@emargolis is still planning to work on it.

emargolis, 2021-07-21:
Was addressed by #7916 . Closing

closed 2021-07-21 by emargolis