project-chip/connectedhomeip #4321: Layer::GetClock_Monotonic() and SystemLayer.StartTimer functions generate inaccurate timing on EFR32

silabs-Simran, 2021-01-11:
The following issue is observed on EFR32 BRD4180A:

**For chip::System::Platform::Layer::GetClock_Monotonic()**, is implemented on CHIP Layer and for some reason, it is not accurate. The codes in the example are expected to send broadcast every 5s, but actually, it's sending every ~1s.

Moreover, the timing is not accurate for **For chip::DeviceLayer::SystemLayer.StartTimer** as well. 

As this is CHIP Layer implementation, I wanted to verify if this is only EFR32 related issue or it affecting on all the platforms and how it can be resolved?

Thank you.

bzbarsky-apple, 2022-06-27:
@mkardous-silabs Any idea what might have been going on here?

@silabs-Simran Is this still an issue?

woody-apple, 2022-07-22:
Issue Review: Per discussion, we can close this, we don't believe this is an issue anymore.

closed 2022-07-22 by woody-apple