project-chip/connectedhomeip #4339: Crash during iOS pairing

woody-apple, 2021-01-12:
#### Problem

Crash found 

Thread 13 name:  Dispatch queue: com.chip.ble.work_queue
Thread 13 Crashed ↩:
0   CHIP                          	0x00000001e608f0e8 chip::Ble::BleLayer::NewBleEndPoint(chip::Ble::BLEEndPoint**, void*, chip::Ble::BleRole, bool) + 52 (BleLayer.cpp:391)
1   CHIP                          	0x00000001e60cbec8 chip::Transport::BLE::InitInternal(void*) + 64 (BLE.cpp:81)
2   CHIP                          	0x00000001e60cbec8 chip::Transport::BLE::InitInternal(void*) + 64 (BLE.cpp:81)
3   CHIP                          	0x00000001e60cc1a0 chip::Transport::BLE::OnBleConnectionComplete(void*, void*) + 44 (BLE.cpp:170)
4   CHIP                          	0x00000001e60b30c4 -[BleConnection peripheral:didDiscoverCharacteristicsForService:error:] + 120 (BleConnectionDelegateImpl.mm:204)
5   CoreBluetooth                 	0x00000001b111a33c -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 224 (CBPeripheral.m:788)
6   CoreBluetooth                 	0x00000001b111a3e8 -[CBPeripheral handleServiceEvent:serviceSelector:delegateSelector:delegateFlag:] + 120 (CBPeripheral.m:798)
7   CoreBluetooth                 	0x00000001b1116c6c -[CBPeripheral handleMsg:args:] + 352 (CBPeripheral.m:229)
8   CoreBluetooth                 	0x00000001b110c2ac -[CBCentralManager handleMsg:args:] + 184 (CBCentralManager.m:1239)
9   CoreBluetooth                 	0x00000001b1134ad0 -[CBManager xpcConnectionDidReceiveMsg:args:] + 204 (CBManager.m:273)
10  CoreBluetooth                 	0x00000001b1127ea4 __30-[CBXpcConnection _handleMsg:]_block_invoke + 64 (CBXpcConnection.m:234)
11  libdispatch.dylib             	0x00000001978c640c _dispatch_call_block_and_release + 24 (init.c:1454)
12  libdispatch.dylib             	0x00000001978c73f4 _dispatch_client_callout + 16 (object.m:559)
13  libdispatch.dylib             	0x000000019786cab8 _dispatch_lane_serial_drain$VARIANT$mp + 656 (inline_internal.h:2560)
14  libdispatch.dylib             	0x000000019786d5e4 _dispatch_lane_invoke$VARIANT$mp + 468 (queue.c:3877)
15  libdispatch.dylib             	0x000000019786c954 _dispatch_lane_serial_drain$VARIANT$mp + 300 (inline_internal.h:2601)
16  libdispatch.dylib             	0x000000019786d5a8 _dispatch_lane_invoke$VARIANT$mp + 408 (queue.c:3877)
17  libdispatch.dylib             	0x00000001978777ac _dispatch_workloop_worker_thread + 772 (queue.c:6634)
18  libsystem_pthread.dylib       	0x00000001e3d4e8bc _pthread_wqthread + 272 (pthread.c:2228)
19  libsystem_pthread.dylib       	0x00000001e3d51b88 start_wqthread + 8

bzbarsky-apple, 2021-01-14:
Line numbers have shifted a bit on tip with rev 9661374ddc013e4101b5a0bdf80d76ce484fae18 but if I assume the stack is from a build that predates that, we are crashing on this line:
```
    *retEndPoint = nullptr;
```
where the caller did:
```
    err = mBleLayer->NewBleEndPoint(&mBleEndPoint, connObj, kBleRole_Central, true);
```
and that first arg is `retEndPoint`.  Which suggests to me that `this` is dead....

woody-apple, 2021-02-02:
```
      6 CHIP: chip::Transport::BLE::OnBleConnectionError(void*, int) <==
        3 CHIP: chip::Transport::BLE::OnBleConnectionComplete(void*, void*)
        | 3 CHIP: chip::Transport::BLE::OnBleConnectionComplete(void*, void*)
        |   3 CHIP: -[BleConnection peripheral:didDiscoverCharacteristicsForService:error:]
        |     3 CoreBluetooth: -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:]
        |       3 CoreBluetooth: -[CBPeripheral handleServiceEvent:serviceSelector:delegateSelector:delegateFlag:]
        |         3 CoreBluetooth: -[CBPeripheral handleMsg:args:]
        |           3 CoreBluetooth: -[CBCentralManager handleMsg:args:]
        |             3 CoreBluetooth: -[CBManager xpcConnectionDidReceiveMsg:args:]
        |               3 CoreBluetooth: __30-[CBXpcConnection _handleMsg:]_block_invoke
        |                 3 libdispatch.dylib: _dispatch_call_block_and_release
        |                   3 libdispatch.dylib: _dispatch_client_callout
        |                     3 libdispatch.dylib: _dispatch_lane_serial_drain
        |                       3 libdispatch.dylib: _dispatch_lane_invoke
        |                         3 libdispatch.dylib: _dispatch_lane_serial_drain
        |                           3 libdispatch.dylib: _dispatch_lane_invoke
        |                             3 libdispatch.dylib: _dispatch_workloop_worker_thread
        |                               3 libsystem_pthread.dylib: _pthread_wqthread
        |                                 3 libsystem_pthread.dylib: 
        3 CHIP: -[BleConnection peripheral:didDiscoverServices:]
          3 CHIP: -[BleConnection peripheral:didDiscoverServices:]
            3 CoreBluetooth: -[CBPeripheral handleServicesDiscovered:]
              3 CoreBluetooth: -[CBPeripheral handleMsg:args:]
                3 CoreBluetooth: -[CBCentralManager handleMsg:args:]
                  3 CoreBluetooth: -[CBManager xpcConnectionDidReceiveMsg:args:]
                    3 CoreBluetooth: __30-[CBXpcConnection _handleMsg:]_block_invoke
                      3 libdispatch.dylib: _dispatch_call_block_and_release
                        3 libdispatch.dylib: _dispatch_client_callout
                          3 libdispatch.dylib: _dispatch_lane_serial_drain
                            3 libdispatch.dylib: _dispatch_lane_invoke
                              3 libdispatch.dylib: _dispatch_lane_serial_drain
                                3 libdispatch.dylib: _dispatch_lane_invoke
                                  3 libdispatch.dylib: _dispatch_workloop_worker_thread
                                    3 libsystem_pthread.dylib: _pthread_wqthread
                                      3 libsystem_pthread.dylib: 
```

vivien-apple, 2021-02-25:
Does it still happens or does #4336 fixed it ? @woody-apple

vivien-apple, 2021-03-09:
The stack for this one and #5193 looks similar too.

The stack is out of context, so it is a bit hard to put the right diagnostic but I wonder if this could be that that someone has tried to connect to a BLE device but decides that it does not want it anymore.
Then the `RendezvousSession` has been released BUT the BLE scanning is still ongoing. And once it finally found something to connect too, it ends up calling the registered callback that point to memory that has been released...

vivien-apple, 2021-03-09:
OK. If I call `DeviceCommissioner::StopPairing` right after a call to `DeviceCommissioner::PairDevice` I can reproduce those stacks:
```
* thread #4, queue = 'com.chip.ble.work_queue', stop reason = EXC_BAD_ACCESS (code=2, address=0x10131c5d4)
  * frame #0: 0x000000010131c5d4
    frame #1: 0x00000001000cd05a chip-tool`chip::Transport::BLE::OnBleConnectionError(appState=0x0000000101312990, err=6001) at BLE.cpp:182:25
    frame #2: 0x00000001000cd00b chip-tool`chip::Transport::BLE::OnBleConnectionComplete(appState=0x0000000101312990, connObj=0x000000010131c800) at BLE.cpp:172:9
    frame #3: 0x00000001000b1023 chip-tool`-[BleConnection peripheral:didDiscoverCharacteristicsForService:error:](self=0x0000000101312fe0, _cmd="peripheral:didDiscoverCharacteristicsForService:error:", peripheral=0x000000010131c800, service=0x00000001010067d0, error=0x0000000000000000) at BleConnectionDelegateImpl.mm:210:5
    frame #4: 0x00007fff2e9c6705 CoreBluetooth`-[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 239
    frame #5: 0x00007fff2e9c67a7 CoreBluetooth`-[CBPeripheral handleServiceEvent:serviceSelector:delegateSelector:delegateFlag:] + 113
    frame #6: 0x00007fff2e9c2b6a CoreBluetooth`-[CBPeripheral handleMsg:args:] + 291
    frame #7: 0x00007fff2e9bd417 CoreBluetooth`-[CBCentralManager handleMsg:args:] + 195
    frame #8: 0x00007fff2e9b4a1a CoreBluetooth`-[CBManager xpcConnectionDidReceiveMsg:args:] + 177
    frame #9: 0x00007fff2e9b495c CoreBluetooth`__30-[CBXpcConnection _handleMsg:]_block_invoke + 50
    frame #10: 0x00007fff68ec66c4 libdispatch.dylib`_dispatch_call_block_and_release + 12
    frame #11: 0x00007fff68ec7658 libdispatch.dylib`_dispatch_client_callout + 8
    frame #12: 0x00007fff68eccc44 libdispatch.dylib`_dispatch_lane_serial_drain + 597
    frame #13: 0x00007fff68ecd609 libdispatch.dylib`_dispatch_lane_invoke + 414
    frame #14: 0x00007fff68eccaf6 libdispatch.dylib`_dispatch_lane_serial_drain + 263
    frame #15: 0x00007fff68ecd5d6 libdispatch.dylib`_dispatch_lane_invoke + 363
    frame #16: 0x00007fff68ed6c09 libdispatch.dylib`_dispatch_workloop_worker_thread + 596
    frame #17: 0x00007fff69121a3d libsystem_pthread.dylib`_pthread_wqthread + 290
    frame #18: 0x00007fff69120b77 libsystem_pthread.dylib`start_wqthread + 15
```

```
 thread #3, queue = 'com.chip.ble.work_queue', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
  * frame #0: 0x00000001000cd1f4 chip-tool`chip::Ble::BLEEndPoint::ConnectionObjectIs(this=0x6369747369726574, connObj=0x00000001010061e0) at BLEEndPoint.h:104:77
    frame #1: 0x00000001000ccf7f chip-tool`chip::Transport::BLE::OnBleConnectionComplete(appState="kCBMsgArgCharacteristicHandle", connObj=0x00000001010061e0) at BLE.cpp:163:5
    frame #2: 0x00000001000b1023 chip-tool`-[BleConnection peripheral:didDiscoverCharacteristicsForService:error:](self=0x0000000100713090, _cmd="peripheral:didDiscoverCharacteristicsForService:error:", peripheral=0x00000001010061e0, service=0x000000010071fc30, error=0x0000000000000000) at BleConnectionDelegateImpl.mm:210:5
    frame #3: 0x00007fff2e9c6705 CoreBluetooth`-[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 239
    frame #4: 0x00007fff2e9c67a7 CoreBluetooth`-[CBPeripheral handleServiceEvent:serviceSelector:delegateSelector:delegateFlag:] + 113
    frame #5: 0x00007fff2e9c2b6a CoreBluetooth`-[CBPeripheral handleMsg:args:] + 291
    frame #6: 0x00007fff2e9bd417 CoreBluetooth`-[CBCentralManager handleMsg:args:] + 195
    frame #7: 0x00007fff2e9b4a1a CoreBluetooth`-[CBManager xpcConnectionDidReceiveMsg:args:] + 177
    frame #8: 0x00007fff2e9b495c CoreBluetooth`__30-[CBXpcConnection _handleMsg:]_block_invoke + 50
    frame #9: 0x00007fff68ec66c4 libdispatch.dylib`_dispatch_call_block_and_release + 12
    frame #10: 0x00007fff68ec7658 libdispatch.dylib`_dispatch_client_callout + 8
    frame #11: 0x00007fff68eccc44 libdispatch.dylib`_dispatch_lane_serial_drain + 597
    frame #12: 0x00007fff68ecd609 libdispatch.dylib`_dispatch_lane_invoke + 414
    frame #13: 0x00007fff68eccaf6 libdispatch.dylib`_dispatch_lane_serial_drain + 263
    frame #14: 0x00007fff68ecd5d6 libdispatch.dylib`_dispatch_lane_invoke + 363
    frame #15: 0x00007fff68ed6c09 libdispatch.dylib`_dispatch_workloop_worker_thread + 596
    frame #16: 0x00007fff69121a3d libsystem_pthread.dylib`_pthread_wqthread + 290
    frame #17: 0x00007fff69120b77 libsystem_pthread.dylib`start_wqthread + 15
```

Will close #5193 as a duplicate and I will do a fix.

PR #5240 (2021-03-19): Add BleConnectionDelegate::CancelConnection to stop scanning when ren…
…dezvous is aborted

 #### Problem

Once `BLEConnectionDelegate::NewConnection` has been called there is no explicit method to stop it. In the case of `darwin`, the BLE scanning continues in the background and it crashes the stack once it the complete or error callback is called...

 #### Summary of Changes
 * Add `BleConnectionDelegate::CancelConnection`
 
 Fixes #4339
closed 2021-03-19 by andy31415