project-chip/connectedhomeip #4341: python chip-device-ctrl fails to connect with raspi on Mac

cjandhyala, 2021-01-13:
python chip-device-ctrl fails to connect with raspi on Mac. Lighting App used.

chip-device-ctrl > ble-scan-connect -t 120 1383
2021-01-12 18:54:55,094 ChipBLEMgr   INFO     adding to scan list:
2021-01-12 18:54:55,094 ChipBLEMgr   INFO
2021-01-12 18:54:55,094 ChipBLEMgr   INFO     Name            = ubuntuserver
2021-01-12 18:54:55,094 ChipBLEMgr   INFO     ID              = C17171C1-7421-4325-AB9C-3F40A8DF8169
2021-01-12 18:54:55,094 ChipBLEMgr   INFO     RSSI            = -59
2021-01-12 18:54:55,094 ChipBLEMgr   INFO     Pairing State   = 0
2021-01-12 18:54:55,094 ChipBLEMgr   INFO     Discriminator   = 1383
2021-01-12 18:54:55,094 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-01-12 18:54:55,094 ChipBLEMgr   INFO     Product Id      = 65279
2021-01-12 18:54:55,094 ChipBLEMgr   INFO     ADV data: {
    kCBAdvDataChannel = 37;
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataLocalName = ubuntuserver;
    kCBAdvDataServiceData =     {
        FEAF = {length = 7, bytes = 0x0067055a23fffe};
    };
    kCBAdvDataServiceUUIDs =     (
        FEAF
    );
}
2021-01-12 18:54:55,094 ChipBLEMgr   INFO
2021-01-12 18:54:55,095 ChipBLEMgr   INFO     scanning stopped
2021-01-12 18:54:55,095 ChipBLEMgr   INFO     trying to connect to 1383
2021-01-12 18:55:10,095 ChipBLEMgr   INFO     connect fail
chip-device-ctrl > 2021-01-12 18:55:10,200 ChipBLEMgr   INFO     BLE disconnected, error = None

erjiaqing, 2021-01-20:
https://github.com/project-chip/connectedhomeip/issues/4399

Since there is another issue related to DevCtrl and BLE, I think these two issue can be fixed together

erjiaqing, 2021-01-21:
Cannot reproduce on my side (running macOS X 10.15.7), please consider restart linux chip-tool-server and chip-device-ctrl on mac.

cjandhyala, 2021-01-23:
tried both i.e. restarting the chip-dev-ctrl and chip-tool-server (tried on linux, raspi and M5) , in either of the cases dev ctrl failed to connect. I am on Mac BigSur

chip-device-ctrl > ble-scan-connect -t 120 CHIP-3480
2021-01-22 16:18:04,419 ChipBLEMgr   INFO     BLE is ready!
2021-01-22 16:18:09,651 ChipBLEMgr   INFO     adding to scan list:
2021-01-22 16:18:09,651 ChipBLEMgr   INFO     
2021-01-22 16:18:09,651 ChipBLEMgr   INFO     Name            = ubuntuserver                                                                    
2021-01-22 16:18:09,651 ChipBLEMgr   INFO     ID              = C17171C1-7421-4325-AB9C-3F40A8DF8169                                            
2021-01-22 16:18:09,651 ChipBLEMgr   INFO     RSSI            = -50                                                                             
2021-01-22 16:18:09,651 ChipBLEMgr   INFO     Pairing State   = 0
2021-01-22 16:18:09,651 ChipBLEMgr   INFO     Discriminator   = 3840
2021-01-22 16:18:09,651 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-01-22 16:18:09,652 ChipBLEMgr   INFO     Product Id      = 65279
2021-01-22 16:18:09,652 ChipBLEMgr   INFO     ADV data: {
    kCBAdvDataChannel = 37;
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataLocalName = ubuntuserver;
    kCBAdvDataServiceData =     {
        FEAF = {length = 7, bytes = 0x00000f5a23fffe};
    };
    kCBAdvDataServiceUUIDs =     (
        FEAF
    );
}
2021-01-22 16:18:09,652 ChipBLEMgr   INFO     
2021-01-22 16:20:04,421 ChipBLEMgr   INFO     scanning stopped
2021-01-22 16:20:04,421 ChipBLEMgr   INFO     trying to connect to CHIP-3480
2021-01-22 16:20:04,421 ChipBLEMgr   INFO     connect fail
chip-device-ctrl > ble-scan-connect -t 120 CHIP-3480
2021-01-22 16:20:51,302 ChipBLEMgr   INFO     adding to scan list:
2021-01-22 16:20:51,302 ChipBLEMgr   INFO     
2021-01-22 16:20:51,303 ChipBLEMgr   INFO     Name            = linuxboxathome                                                                  
2021-01-22 16:20:51,303 ChipBLEMgr   INFO     ID              = B44C1B0B-3B1B-46E4-9699-87BBE6D53759                                            
2021-01-22 16:20:51,303 ChipBLEMgr   INFO     RSSI            = -52                                                                             
2021-01-22 16:20:51,303 ChipBLEMgr   INFO     Pairing State   = 0
2021-01-22 16:20:51,303 ChipBLEMgr   INFO     Discriminator   = 3840
2021-01-22 16:20:51,303 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-01-22 16:20:51,303 ChipBLEMgr   INFO     Product Id      = 65279
2021-01-22 16:20:51,303 ChipBLEMgr   INFO     ADV data: {
    kCBAdvDataChannel = 37;
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataLocalName = linuxboxathome;
    kCBAdvDataServiceData =     {
        FEAF = {length = 7, bytes = 0x00000f5a23fffe};
    };
    kCBAdvDataServiceUUIDs =     (
        FEAF
    );
}
2021-01-22 16:20:51,303 ChipBLEMgr   INFO     
2021-01-22 16:22:50,526 ChipBLEMgr   INFO     scanning stopped
2021-01-22 16:22:50,526 ChipBLEMgr   INFO     trying to connect to CHIP-3480
2021-01-22 16:22:50,527 ChipBLEMgr   INFO     connect fail
chip-device-ctrl > ble-scan-connect -t 120 CHIP-3480
2021-01-22 16:51:29,229 ChipBLEMgr   INFO     adding to scan list:
2021-01-22 16:51:29,229 ChipBLEMgr   INFO     
2021-01-22 16:51:29,229 ChipBLEMgr   INFO     Name            = CHIP-3840                                                                       
2021-01-22 16:51:29,230 ChipBLEMgr   INFO     ID              = E27D6633-2301-4061-A5C9-1401ED6F864F                                            
2021-01-22 16:51:29,230 ChipBLEMgr   INFO     RSSI            = -44                                                                             
2021-01-22 16:51:29,232 ChipBLEMgr   INFO     Pairing State   = 0
2021-01-22 16:51:29,232 ChipBLEMgr   INFO     Discriminator   = 3840
2021-01-22 16:51:29,232 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-01-22 16:51:29,232 ChipBLEMgr   INFO     Product Id      = 65279
2021-01-22 16:51:29,233 ChipBLEMgr   INFO     ADV data: {
    kCBAdvDataChannel = 37;
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataLocalName = "CHIP-3840";
    kCBAdvDataServiceData =     {
        FEAF = {length = 7, bytes = 0x00000f5a23fffe};
    };
}
2021-01-22 16:51:29,233 ChipBLEMgr   INFO     
2021-01-22 16:53:27,228 ChipBLEMgr   INFO     scanning stopped
2021-01-22 16:53:27,230 ChipBLEMgr   INFO     trying to connect to CHIP-3480
2021-01-22 16:53:27,230 ChipBLEMgr   INFO     connect fail

yunhanw-google, 2021-01-28:
Hi, Chaitanya

I just reproduce your issue again with ble dongle

here is my observation and analysis

I attach external ble dongle, reboot, it shows
ubuntu@ubuntu:~/connectedhomeip$ hciconfig
hci0:	Type: Primary  Bus: USB
	BD Address: 5C:F3:70:8A:0E:F0  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN
	RX bytes:2096 acl:19 sco:0 events:108 errors:0
	TX bytes:3942 acl:14 sco:0 commands:98 errors:0

which has hide the internal ble chip
python device controller in mac cannott make tthe ble cconnection to raspi4
then
i extract ble dongle, and reboot
it shows
ubuntu@ubuntu:~$ hciconfig hci0:	Type: Primary Bus: UART BD Address: DC:A6:32:8D:E0:57 ACL MTU: 1021:8 SCO MTU: 64:1 UP RUNNING PSCAN ISCAN RX bytes:1583 acl:0 sco:0 events:99 errors:0 TX bytes:2105 acl:0 sco:0 commands:99 errors:0
internal uart ble chip comes back
python device controller can make successful ble connection to raspi4 now
therefore
the root cause is that
once the external ble dongle is attached, ubuntu 20.04 masks the internal ble chip and show only usb bluetooth interface, which would confuse bluetooth-related connection from external

closed 2021-01-28 by yunhanw-google