project-chip/connectedhomeip #4356: Crash during iOS Pairing

woody-apple, 2021-01-14:
#### Problem
Thread 5 Crashed:
0   CHIP                          	0x0000000102773610 chip::RendezvousSession::UpdateState(chip::RendezvousSession::State) + 72 (RendezvousSession.cpp:266)
1   CHIP                          	0x0000000102773a10 chip::RendezvousSession::OnRendezvousError(int) + 180 (RendezvousSession.cpp:258)
2   CHIP                          	0x0000000102773a10 chip::RendezvousSession::OnRendezvousError(int) + 180 (RendezvousSession.cpp:258)
3   CHIP                          	0x0000000102778af0 chip::Transport::BLE::OnBleEndPointConnectionClosed(chip::Ble::BLEEndPoint*, int) + 88 (BLE.cpp:230)
4   CHIP                          	0x000000010273c63c chip::Ble::BLEEndPoint::DoCloseCallback(unsigned char, unsigned char, int) + 116 (BLEEndPoint.cpp:454)
5   CHIP                          	0x000000010273c4b4 chip::Ble::BLEEndPoint::FinalizeClose(unsigned char, unsigned char, int) + 120 (BLEEndPoint.cpp:392)
6   CHIP                          	0x000000010273bbe4 chip::Ble::BLEEndPoint::DoClose(unsigned char, int) + 168 (BLEEndPoint.cpp:351)
7   CHIP                          	0x000000010273e1f8 chip::Ble::BLEEndPoint::HandleAckReceivedTimeout(chip::System::Layer*, void*, int) + 152 (BLEEndPoint.cpp:1704)
8   CHIP                          	0x00000001027710b8 chip::System::Timer::HandleComplete() + 252 (SystemTimer.cpp:274)
9   CHIP                          	0x000000010276f674 chip::System::Layer::HandleSelectResult(int, fd_set*, fd_set*, fd_set*) + 456 (SystemLayer.cpp:698)
10  CHIP                          	0x000000010275bd40 chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::SysProcess() + 260 (GenericPlatformManagerImpl_POSIX.cpp:170)
11  CHIP                          	0x000000010275bad8 chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() + 52 (GenericPlatformManagerImpl_POSIX.cpp:194)
12  CHIP                          	0x000000010275c0c4 chip::DeviceLayer::PlatformManager::RunEventLoop() + 24 (PlatformManager.h:208)
13  CHIP                          	0x000000010275bed8 chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) + 48 (GenericPlatformManagerImpl_POSIX.cpp:205)
14  libsystem_pthread.dylib       	0x00000001da092aa4 _pthread_start + 288 (pthread.c:881)
15  libsystem_pthread.dylib       	0x00000001da097874 thread_start + 8

bzbarsky-apple, 2021-01-14:
I _think_ this is crashing on:
```
    if (mDelegate != nullptr)
```
which would suggest that `this` (the `RendezvousSession`) is dead....

vivien-apple, 2021-01-15:
I wonder if this is related to https://github.com/project-chip/connectedhomeip/pull/4336. This is hard to say without steps to reproduce :'(

woody-apple, 2021-02-02:
Note: this is pulled from a crash aggregator. Definitely showing up commonly-ish. Marking as a P1.

vivien-apple, 2021-02-25:
Does it still happens or does #4336 fixed it ? @woody-apple

vivien-apple, 2021-02-25:
Seems like it does still happens looking at #4965

vivien-apple, 2021-03-05:
Pretty sure that #5156 should have fixed it. Feel free to reopen if it still happens.

closed 2021-03-05 by vivien-apple