project-chip/connectedhomeip #4366: Add ARM64 protoc into CIPD to avoid Manual installation of protoc when setting up CHIP env on Raspi running Linux

SaileeG, 2021-01-14:
Initially bootstrap worked because the packages it needs (python, gn) were available on arm64
Now bootstrap needs python, gn, and protoc where protoc is not installed by default during the env setup causing https://github.com/project-chip/connectedhomeip/issues/4315 and need to be installed separately (sudo apt-get install protobuf-compiler).

This ticket is to track addition of ARM64 protoc into CIPD to avoid this additional step.

mohrr, 2021-01-14:
This should be fixed by https://pigweed-review.googlesource.com/c/pigweed/pigweed/+/29489 which switches to a protoc package that's built for more platforms (and kept more up-to-date). It'll need to be submitted to Pigweed and then CHIP's version of Pigweed will need to be updated.

mspang, 2021-02-02:
Already fixed.

closed 2021-02-02 by mspang