project-chip/connectedhomeip #4369: Fix GENERATED_FUNCTION array in endpoint config template

jepenven-silabs, 2021-01-14:
#### Problem
Appbuilder, the predecessor of ZAP was a All in one application which could generate code for both the ZCL and the actual implementation. With ZAP the only information available is the one provided by the ZCL8 spec. Meaning that ZAP cannot produce code for things that he's not aware of. For example the fact that some cluster have an `Init` function and some doesn't have any. As a consequence the `GENERATED_FUNCTION` array is now being generated with a hardcoded preset since ZAP is unable to produce it.

 #### Proposed Solution
Find a way to either : 
Removed the `GENERATED_FUNCTIONS` array by changing the ZCL framework
Add back the `GENERATED_FUNCTIONS` array in a way that is easily scalable and not hardcoded directly in the `helper.js` file. Meaning that we should have an external way of signifying to zap if a cluster have an `INIT_FUNCTION`, `MESSAGE_SENT_FUNCTION` or  `PRE_ATTRIBUTE_CHANGED_FUNCTION`

woody-apple, 2022-02-08:
It is not clear at this point, as everything behind it is quite out of date. @jepenven-silabs Can you please clarify what's needed now?

bzbarsky-apple, 2023-02-09:
This is very much still an issue.  We are moving the lists of things into a YAML, which may help a bit, but it's still a hardcoded list in the SDK...