project-chip/connectedhomeip #4371: Investigate address and memory sanitize errors when building all_clusters_app for linux

andy31415, 2021-01-14:
#### Problem
 
 Memory sanitize complains about glib, address sanitize gives a potential bug:
 
 ```
==2151032==ERROR: AddressSanitizer: global-buffer-overflow on address 0x55a9bb9e15d6 at pc 0x55a9bae1235f bp 0x7ffcf4e91c10 sp 0x7ffcf4e913d8
READ of size 8 at 0x55a9bb9e15d6 thread T0
    #0 0x55a9bae1235e in __asan_memmove (/usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/all-clusters-server+0x11135e)
    #1 0x55a9bae6e47f in typeSensitiveMemCopy(unsigned char*, unsigned char*, EmberAfAttributeMetadata*, bool, unsigned short) /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/app/util/attribute-storage.cpp:394:13
    #2 0x55a9bae6d785 in emAfReadOrWriteAttribute(EmberAfAttributeSearchRecord*, EmberAfAttributeMetadata**, unsigned char*, unsigned short, bool) /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/app/util/attribute-storage.cpp:563:47
    #3 0x55a9bae71517 in emAfReadAttribute(unsigned char, unsigned short, unsigned short, unsigned char, unsigned short, unsigned char*, unsigned short, unsigned char*) /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/app/util/attribute-table.cpp:645:31
    #4 0x55a9bae60c71 in emberAfPluginReportingTickEventHandler() /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/app/reporting/reporting.cpp:220:18
    #5 0x55a9bae68c9c in EventControlHandler(chip::System::Layer*, void*, int) /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/app/util/af-event.cpp:109:17
    #6 0x55a9baeb8377 in chip::System::Timer::HandleComplete() /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/system/SystemTimer.cpp:274:9
    #7 0x55a9baeb45f0 in chip::System::Layer::HandleSelectResult(int, fd_set*, fd_set*, fd_set*) /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/system/SystemLayer.cpp:698:21
    #8 0x55a9bae97584 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::SysProcess() /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_POSIX.cpp:170:21
    #9 0x55a9bae96f65 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_POSIX.cpp:194:9
    #10 0x55a9bae3eb57 in chip::DeviceLayer::PlatformManager::RunEventLoop() /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../third_party/connectedhomeip/src/include/platform/PlatformManager.h:204:37
    #11 0x55a9bae3ea9e in main /usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/../main.cpp:66:38
    #12 0x7f73b6bdad09 in __libc_start_main csu/../csu/libc-start.c:308:16
    #13 0x55a9bad99149 in _start (/usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/all-clusters-server+0x98149)

0x55a9bb9e15d6 is located 42 bytes to the left of global variable 'singletonAttributeData' defined in '../third_party/connectedhomeip/src/app/util/attribute-storage.cpp:67:9' (0x55a9bb9e1600) of size 6
0x55a9bb9e15d6 is located 0 bytes to the right of global variable 'attributeData' defined in '../third_party/connectedhomeip/src/app/util/attribute-storage.cpp:60:9' (0x55a9bb9e1540) of size 150
SUMMARY: AddressSanitizer: global-buffer-overflow (/usr/local/google/home/andreilitvin/devel/connectedhomeip/examples/all-clusters-app/linux/out/all-clusters-server+0x11135e) in __asan_memmove
Shadow bytes around the buggy address:
  0x0ab5b7734260: f9 f9 f9 f9 00 f9 f9 f9 f9 f9 f9 f9 00 f9 f9 f9
  0x0ab5b7734270: f9 f9 f9 f9 00 f9 f9 f9 f9 f9 f9 f9 00 f9 f9 f9
  0x0ab5b7734280: f9 f9 f9 f9 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab5b7734290: 00 00 00 00 00 00 00 00 f9 f9 f9 f9 00 00 00 00
  0x0ab5b77342a0: 00 00 f9 f9 f9 f9 f9 f9 00 00 00 00 00 00 00 00
=>0x0ab5b77342b0: 00 00 00 00 00 00 00 00 00 00[06]f9 f9 f9 f9 f9
  0x0ab5b77342c0: 06 f9 f9 f9 f9 f9 f9 f9 01 f9 f9 f9 f9 f9 f9 f9
  0x0ab5b77342d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab5b77342e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f9 f9
  0x0ab5b77342f0: f9 f9 f9 f9 00 f9 f9 f9 f9 f9 f9 f9 02 f9 f9 f9
  0x0ab5b7734300: f9 f9 f9 f9 00 f9 f9 f9 f9 f9 f9 f9 00 f9 f9 f9
  ```

 #### Proposed Solution
Determine if this is a real out of bounds and fix.

andy31415, 2021-01-14:
It seems like a real bug:
ATTRIBUTE_MAX_SIZE is 150
and the code attempts to copy 8 bytes from index 147 when I am running through it.

I do not yet understand the attribute storage so finding out where the error lies (is this an invalid index or invalid allocated size) will take a while.

andy31415, 2021-01-14:
@vivien-apple - do you know what defines ATTRIBUTE_MAX_SIZE? should we have additional checks to not allow out of bounds access? I found following attribute-storage.cpp quite hard due to very deep for/if nesting and ternary operator usage.

andy31415, 2021-01-14:
Adding a check results in a barrage of:

```
CHIP:ZCL: READ: attribute size too large for caller
CHIP:ZCL: ERR: reading cluster 0x 6 attribute 0x 0: 0x89
```

Which seems good since this is where the code would have crashed. Guessing minimal storage size we need is 155 bytes (but how does one keep track of that ?!?)

bzbarsky-apple, 2021-01-14:
The attribute table and `ATTRIBUTE_MAX_SIZE` are both correct as far as that goes.

So reporting is doing:
```
        status = emAfReadAttribute(entry.endpoint, entry.clusterId, entry.attributeId, entry.mask, entry.manufacturerCode,
                                   (uint8_t *) &readData, READ_DATA_SIZE, &dataType);
```
where `READ_DATA_SIZE` is defined like so:
```
#ifdef ATTRIBUTE_LARGEST
#define READ_DATA_SIZE ATTRIBUTE_LARGEST
#else
#define READ_DATA_SIZE 8 // max size if attributes aren't present
#endif
```
and `examples/all-clusters-app/all-clusters-common/gen/endpoint_config.h` has:
```
#define ATTRIBUTE_LARGEST (8)
```
because some attributes involved are in fact 8 bytes (e.g. some of the IEEE address stuff in IAS Zone).  This all makes sense in that it's providing a buffer of size `READ_DATA_SIZE`, fine.

Now `emAfReadAttribute` just passes that buffer+length pair to `emAfReadOrWriteAttribute` which locates the right attribute and does `typeSensitiveMemCopy`, passing it the place where the data is coming from, the destination, the attribute descriptor, and the size of the destination buffer.

Here is where things go off the rails.  Quoting `typeSensitiveMemCopy`:
```
static EmberAfStatus typeSensitiveMemCopy(uint8_t * dest, uint8_t * src, EmberAfAttributeMetadata * am, bool write,
                                          uint16_t readLength)
...
    uint16_t size                      = (readLength == 0) ? am->size : readLength;
...
        if (!write && readLength != 0 && readLength < am->size)
        {
            return EMBER_ZCL_STATUS_INSUFFICIENT_SPACE;
        }
...            memmove(dest, src, size);
```
So we make sure we are not _writing_ out of bounds if the caller buffer is not big enough.  But we will happily _read_ out of bounds.

bzbarsky-apple, 2021-01-14:
OK, so the intended semantics here are documented on `emAfReadOrWriteAttribute` as follows:
```
// When reading non-string attributes, this function returns an error when destination
// buffer isn't large enough to accommodate the attribute type.  For strings, the
// function will copy at most readLength bytes.  This means the resulting string
// may be truncated.  The length byte(s) in the resulting string will reflect
// any truncation.  If readLength is zero, we are working with backwards-
// compatibility wrapper functions and we just cross our fingers and hope for
// the best.
//
// When writing attributes, readLength is ignored.  For non-string attributes,
// this function assumes the source buffer is the same size as the attribute
// type.  For strings, the function will copy as many bytes as will fit in the
// attribute.  This means the resulting string may be truncated.  The length
// byte(s) in the resulting string will reflect any truncated.
```
So there are a few cases:
* Reading a string: Should copy as much as we can into a buffer of size readLength.  This part seems to work right.... as long as `readLength` (the size of the buffer we are reading into) is at least 1 for a normal string or 2 for a long string; otherwise it goes off the rails, imo (ends up with a very-large size that gets passed to `emberAfCopy(Long)String`).
* Reading a non-string: Should fail if destination size is known and is smaller that the size of the attribute.  This part works right.
* Reading a non-string: Should read the actual amount of bytes the attribute takes up.  This is broken: we read the size of the passed-in buffer.
* Writing a string: The bit about `readLength` being ignored is a lie at first glance, _but_ `emAfWriteAttribute` actually passes 0 for `readLength`, so it ends up being kinda true, because if `readLength` is 0 its value is ignored.  With that caveat, we end up writing the right thing, I think.
* Writing a non-string: Again, `readLength` being ignored is not really true, but ends up true in practice because caller always passes 0.  We end up writing a number of bytes equal to the attribute size, which is what the comments claim should happen.

@selissia am I missing anything here?

bzbarsky-apple, 2021-01-14:
I have some thoughts on fixing this; will poke at it in a few hours.

vivien-apple, 2021-01-15:
> @vivien-apple - do you know what defines ATTRIBUTE_MAX_SIZE? should we have additional checks to not allow out of bounds access? I found following attribute-storage.cpp quite hard due to very deep for/if nesting and ternary operator usage.

This the total size of all attributes inside the `GENERATED_ATTRIBUTES` macro.

So with ZAP `ATTRIBUTE_MAX_SIZE` represents this value: https://github.com/project-chip/zap/blob/8312cf7d8a17279ef16825c65db8bba0bcb64696/src-electron/generator/helper-endpointconfig.js#L424

andy31415, 2021-01-15:
@bzbarsky-apple, @selissia - the fact that we had to list cases for read/write for that method and they are complex has me worried about code maintainability. 

I would also add for non-strings, I currently seem to see we always allocate 8-bytes of buffer (presumably size of 64-bit int which is the largest) and we read into it. Reading shorter data would work on littleendian, but maybe bigendian would fail. I don't think we generally have big endian devices, however it would still feel more robust if we could ensure sizes always match.

closed 2021-01-15 by andy31415