project-chip/connectedhomeip #4372: Sending PacketBuffer with "ref count > 1" fails in LwIP

pan-apple, 2021-01-14:
#### Problem
Sending a message in a `PacketBuffer` with reference count > 1 causes an assert in LwIP library. 
```
assertion "p->ref == 1" failed: file "../esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c", line 927, function: ip4_output_if_opt_src
```

Similar checks are in IPv6 code as well.

CRMP uses reference count as a mechanism to cache the transmitted packets for future retransmissions. But, given the above checks, it won't work on platforms that are using LwIP.

 #### Proposed Solution
One option is to deep copy the packets to another PacketBuffer before transmission, instead of incrementing the reference count. This does add memory overheads.

Need to further analyze the issue to identify the solution.

yufengwangca, 2021-01-25:
When retaining a buffer with CRMP, the caller expects the msg to be unmodified.  LwIP stack will normally prepend the packet headers as the packet traverses the UDP/IP/netif layers, which normally modifies the packet.  

We need to deep-copies msg into a fresh object, and queues that for transmission, leaving the original msg available after return.

bzbarsky-apple, 2021-01-29:
It sounds like LwIP is asserting that we have handed it unique ownership of the packet buffer, right?

If that's the case, the only real option I can see is that when saving a copy for CRMP purposes we actually make a copy instead of refcounting.  But if we're doing that, why do we need a "is retained buffer" flag on the message header, as #4571 proposes?  How would it be used?

yufengwangca, 2021-01-29:
Refactor CRMP to support both retained buffer and copied buffer is complicated. Actually, this is a well known issue and see how this is addressed in weave: 
https://github.com/openweave/openweave-core/blob/master/src/inet/UDPEndPoint.cpp#L519

We used to have this change in CHIP, but get cleaned up later.

bzbarsky-apple, 2021-01-29:
> Refactor CRMP to support both retained buffer and copied buffer is complicated

Why?  The current way CRMP works in CHIP is that it hands a PacketBuffer off to someone and gives the someone a slot to retain things into.  The someone then does various encryption and whatnot and writes to that retain slot.  Specifically, `SecureSessionMgr::SendMessage` does:
```
    if (bufferRetainSlot != nullptr)
    {
        encryptedMsg        = msgBuf.Retain();
...
        (*bufferRetainSlot) = std::move(encryptedMsg);
```
The only change we need to make, as far as I can see, is that if `bufferRetainSlot` is non-null we create a clone of `msgBuf` into `encryptedMsg`, instead of calling `Retain`.  Why does that not work?  What am I missing?

I know how it's addressed in Weave.  Weave has a quite different packet buffer ownership model than CHIP, fwiw.

bzbarsky-apple, 2021-01-29:
I guess one issue is that at the point when we retain right now the PacketHeader has not been written into the buffer yet?  It seems like we could clone-and-write-it, if that's the issue?

pan-apple, 2021-01-30:
I was actually thinking of moving PacketHeader encoding decoding to SecureMessageCodec. That'll mean that SecureSessionMgr::SendMsg will have a complete packet before it hands it over to the raw transport (UDP). It can copy the packet at that point.

bzbarsky-apple, 2021-01-30:
Yes, we could do that too.  I guess we'd have a separate "insecure message codec" that can be used to send things like the PASE/CASE session establishment messages and would also encode the PacketHeader?

yufengwangca, 2021-02-02:
@bzbarsky-apple 

Right now, the sequence is:
1. retain the buffer
2. send the message (buffer reference count -1)
3. rewind the buffer

Note, we need to rewind the buffer after sending it out, if we use Copy instead of Retain, we need to revisit this logic, right?

Inet layer is used to abstract the various of low level transports, the logic in messaging layer should remain same regardless of the TCP/IP stacks it is running over.

bzbarsky-apple, 2021-02-02:
> Note, we need to rewind the buffer after sending it out, if we use Copy instead of Retain, we need to revisit this logic, right?

Yes, if we copy we might need to do the rewinding differently (or not at all!), depending on which exact things the rewinding is handling.

> The logic in messaging layer should remain same regardless of the TCP/IP stacks it is running over.

Yes, that's fine.  My proposal above is transport-agnostic, right?

yufengwangca, 2021-02-02:
> Yes, that's fine. My proposal above is transport-agnostic, right?

Do you means we switch to clone buffer from retain buffer?  I thought we need to keep the retain buffer method for regular TCP/IP stack, and handle Lwip as a special case.

bzbarsky-apple, 2021-02-02:
> Do you means we switch to clone buffer from retain buffer?

bzbarsky-apple, 2021-02-02:
> Do you means we switch to clone buffer from retain buffer? 

Yes, that is my proposal.

> I thought we need to keep the retain buffer method for regular TCP/IP stack, and handle Lwip as a special case.

TCP is not relevant here, to be clear; what matters is UDP/IP.

We could do a thing where we retain on top of some UDP implementations but end up cloning for others, as an optimization for the non-cloning ones.  What happens in practice?  Which stack will what sort of devices actually use?  If in practice we're cloning on all the really small devices (because they use LwIP), what's the use of trying to avoid cloning on larger devices that use some other IP stack?

yufengwangca, 2021-02-03:
@robszewczyk @turon 
Since copy buffer method always works, I would like to have your view on this, is LwIP equivalent to small devices? I remember there are several deeply embedded devices do not use LwIP, this prevent us from using copy buffer method in CRMP.

yufengwangca, 2021-02-04:
@bzbarsky-apple 

We have internal discussion regarding your proposal, we suggest to keep current implementation of CRMP since we can not assume all small devices use LwIP,  I are checking how many our internal embedded products do not use LwIP now, but there will eventually be non-lwIP small embedded.

Inet is the UDP/IP abstraction layer which deal with the facts before the packets reach a low level UDP/IP stack API, so I suggest to do the clone over there as Weave does.

bzbarsky-apple, 2021-02-04:
OK, so if we posit that non-LwIP small devices exist and the cost of cloning is not acceptable there, I see three options, ranked roughly in order in which I think I'd prefer them:

1. Just add an LwIP ifdef in SSM for the clone-or-retain behavior.  This is the simplest approach in some ways: it minimizes the moving parts that need to interact, and restricts the complexity to the one place that really cares about it: the CRMP bits in SSM.  There's a concern that this is a layering violation, but note that we already have this sort of ifdef in SecureMessageCodec, for example.
1. Modify the transport API so that there are both rvalue-ref and const-lvalue-ref versions of `SendMessage` (on transport and transport manager).  Callers that are actually retaining would call the latter version.  This moves the "I am retaining" signal more in-band with the actual call into the transport layer, as long as the retain happens just above transport.  Which it does, because it happens in the `SecureSessionMgr`.
1. Do the thing Weave does: add a flag that you pass along with the packet indicating that even through you look like you are handing over ownership that is actually a lie and you retained it up the callstack and hence the callee should copy if it wants to.  This would presumably then mean an LwIP ifdef in the UDP protocol code where that flag is checked, etc.

yufengwangca, 2021-02-18:
@bzbarsky-apple 
Sorry for the late response, I am busy on something else.

I try to implement the solution 1 as you suggested, but it seems this route is not easy as we thought, simply clone the PacketBufferHandle is not enough since the transport always re-attach the packet header on top of it, we also need to figure out how to clone the packet header in addition to the packet buffer.

I created a draft PR for your reference. It is still not working, but I would like to pause here until we make decision.
https://github.com/project-chip/connectedhomeip/pull/4927

It looks the solution 3(As weave did) is the easiest route .

closed 2021-03-16 by bzbarsky-apple