project-chip/connectedhomeip #4378: Disable commisionable discovery when it is turned on

arunbharadwaj, 2021-01-14:
#### Problem
Currently, when a device is advertising in "commisionable" mode, the advertising would continue even after the device is commisioned. We need to provide a way for the device maker to disable the mode after a device is commisioned (or any other time)

 #### Proposed Solution
Commisionable mode advertisement is started by calling the Advertise() routine in the mdns library. To stop advertising, we can either:

- Advertise "nothing"
- Completely stop advertising and cleanup listening sockets. To start advertising again, we will have to start listening on the socket again.

woody-apple, 2021-02-02:
@pan-apple Is this complete now?

pan-apple, 2021-02-04:
This same issue was there for BLE based commissioning window. That part of the issue is fixed now. The DNS-SD based advertisement issue still persists. Likely, we can use the same approach to restrict the pairing window in this case.

bzbarsky-apple, 2022-06-27:
This is fixed in the sense that the commissioning window manager stops the advertising.  How long DNS-SD things persist will depend on TTLs, but we should also be doing updates at this point to withdraw advertisements in practice....