project-chip/connectedhomeip #4392: Groups cluster mutations need to affect what addresses the node listens on

bzbarsky-apple, 2021-01-15:
#### Problem
The way group-based multicast works in the CHIP spec is by using an IPv6 multicast address that incorporates the group ID.

This means that a node must listen on all the multicast addresses that correspond to groups listed in the groups clusters on all its endpoints.  When entries are added or removed, the set of addresses being listened to may need to change.

 #### Proposed Solution
This might need to be several issues, but it seems like we need to:
1. Add a callback the groups cluster calls when the set of groups on an endpoint is mutated.
2. Add some APIs to examine the group table and determine what set of addresses we should be listening on.
3. Add the actual code to start/stop listening on the relevant addresses as needed.

@pan-apple @rwalker-apple @andy31415

yufengwangca, 2022-02-02:
This is a missing feature we need to do for group messaging. 
@rcasallas-silabs Do we have the ETA for this feature?

jepenven-silabs, 2022-02-08:
Done in Multiple PR. 
#12936
#11368
#12342 etc..