project-chip/connectedhomeip #4397: Update build system for ember and decouple from generated files

andy31415, 2021-01-15:
#### Problem

code within src/app is not separated by library (e.g. having `src/app/util/BUILD.gn`) and cannot be built stand-alone because it references generated files.

This makes it hard to build unit tests for individual functionality and layering architecture is not checked/enforced by the build system.

 #### Proposed Solution
- stop depending on generated files
- separate into individual static libraries and add some initial unit tests (so that if we ever find bugs or update the code, we can add more unit tests and ensure no regressions)

andy31415, 2021-01-15:
Some thoughts:
  - I expect generated code to depend in app/..., never vice-versa
  - I see a lot of commented out includes inside app, we should probably remove those
  - some cleanup passes would be nice (e.g. convert to `#pragma once` for headers, use C++ things like `constexpr`, add more const correctness).

bzbarsky-apple, 2021-01-15:
See https://github.com/project-chip/connectedhomeip/issues/1630 but the short story is that we don't have a good solution for this yet: Silabs code leans _very_ heavily on the generator and the core code uses data structures that are sized based on the generator's input and whatnot.  We've poked at this a bit, but without success so far.

The simplest way of creating a unit test in the short to medium term is probably to create a unit test app...

There's definitely a lot of cleanup to be done one these files; it's just that we haven't gotten to it yet.

@rwalker-apple @vivien-apple

andy31415, 2021-01-18:
Maybe we could incrementally move things that do not depend on generated code as a stand-alone.

Unfortunately I am not sure how viable because it seems many things are in `af.h` to start with rather than separated (like methods implemented in util.cpp are declared in af.h rather than util.h). Maybe we could have some planning discussion to see what can be done.