project-chip/connectedhomeip #4406: Persistent Storage: move example into a unit test

rgoliver, 2021-01-18:
#### Problem
Reference: https://github.com/project-chip/connectedhomeip/pull/4367

The KVS test is currently a self contained example application, this would be better as a unit test but currently EFR32 does not have a test runner.

 #### Proposed Solution
Once all platforms with Persistent Storage have a test runner, move the example into a unit test.

kkasperczyk-no, 2021-02-10:
@rgoliver I have implemented KeyValueStoreManagerImpl on nrfconnect and I need test added to the unit tests to verify if my implementation works well, as I don't plan adding persistent-storage example for nrfconnect. Are you ok with that I will copy example to the unit tests and after all platforms will have test runner, original example will be removed?

rgoliver, 2021-02-10:
> @rgoliver I have implemented KeyValueStoreManagerImpl on nrfconnect and I need test added to the unit tests to verify if my implementation works well, as I don't plan adding persistent-storage [...]

Sounds great, that definitely moves us in the right direction.

closed 2022-10-09 by stale[bot]