project-chip/connectedhomeip #4424: Not able to build on raspberry pi 4

somshekharjuja, 2021-01-20:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior 
  With the build procedure mentioned, we should be able to fully build chip in linux based raspberry pi 4. 
 * actual behavior
  When we execute "source scripts/activate.sh" or source scripts/bootstrap.sh, there is an error downloading and installing 
  packages. Below are the logs 👍 

Setting up CIPD package manager...[\]Failed to initialize CIPD. Run `CIPD_HTTP_USER_AGENT_PREFIX=pigweed-infra/tools/5a4db08b7b1df9261f5e2491ccb4d0caa10e6bb3/manual /home/pi/connectedhomeip/.environment/cipd/cipd selfupdate -version-file '/home/pi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/.cipd_version'` to diagnose if this is persistent.
Traceback (most recent call last):
  File "/home/pi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 593, in <module>
    sys.exit(main())
  File "/home/pi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 585, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/pi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 307, in setup
    result = step()
  File "/home/pi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 368, in cipd
    cipd_client = cipd_wrapper.init(install_dir, silent=True)
  File "/home/pi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 298, in init
    bootstrap(client, silent)
  File "/home/pi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 262, in bootstrap
    tmp.write(client_bytes())
  File "/home/pi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 246, in client_bytes
    raise Exception('failed to download client')
Exception: failed to download client
Error during bootstrap--see messages above.

 * steps to reproduce
   I have followed this building guide to build but I see that there is an error when we execute source scripts/bootstrap.sh or 
   source scripts/activate.sh. 
    https://github.com/project-chip/connectedhomeip/blob/master/docs/BUILDING.md 
 * system configuration
    Raspberry pi 4 running raspberry-lite image with details mentioned below
    pi@raspberrypi:~/connectedhomeip $ uname -a
    Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

somshekharjuja, 2021-01-20:
pi@raspberrypi:~/connectedhomeip $ git log
commit b430c424ab62438fffae71c9bd2f23d3f0d442d2 (HEAD -> master, origin/master, origin/HEAD)
Author: Rob Walker <robert_g_walker@apple.com>
Date:   Thu Jan 14 15:03:14 2021 -0800

    ignore submodules in restyle_diff.sh

and repo is upto date as you can see above.

somshekharjuja, 2021-01-21:
restyle-diff.sh already has this ignore submodule line but even then I am not able successfullly run activate.sh/bootstarp.sh

Please help

erjiaqing, 2021-01-21:
You should run arm64 system (like ubuntu server for arm64)

woody-apple, 2021-02-03:
Agree, this seems to be building for me. Are you able to confirm that this fixes it?

closed 2021-02-03 by woody-apple