project-chip/connectedhomeip #4443: Unable to build and run chip-tool on Raspberry Pi

Ramin2094, 2021-01-21:
#### Problem

 * expected behavior
chip-tool runs on Raspberry Pi like on a UBUNTO machine 

 * actual behavior
After running:
```
cd examples/chip-tool
git submodule update --init
source third_party/connectedhomeip/scripts/activate.sh
```

I get following error:
```
Setting up CIPD package manager...[\]Failed to initialize CIPD. Run `CIPD_HTTP_USER_AGENT_PREFIX=pigweed-infra/tools/019974b44a68c9ef6dd5845fc1578b0c64156ce0/manual /home/pi/Desktop/CHIP/connectedhomeip/examples/chip-tool/third_party/connectedhomeip/.environment/cipd/cipd selfupdate -version-file '/home/pi/Desktop/CHIP/connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/.cipd_version'` to diagnose if this is persistent.                                                        Traceback (most recent call last):
  File "/home/pi/Desktop/CHIP/connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 593, in <module>
    sys.exit(main())
  File "/home/pi/Desktop/CHIP/connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 585, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/pi/Desktop/CHIP/connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 307, in setup
    result = step()
  File "/home/pi/Desktop/CHIP/connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 368, in cipd
    cipd_client = cipd_wrapper.init(install_dir, silent=True)
  File "/home/pi/Desktop/CHIP/connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 298, in init
    bootstrap(client, silent)
  File "/home/pi/Desktop/CHIP/connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 262, in bootstrap
    tmp.write(client_bytes())
  File "/home/pi/Desktop/CHIP/connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 246, in client_bytes
    raise Exception('failed to download client')
Exception: failed to download client
Error during bootstrap--see messages above.
```

 * steps to reproduce
Follow the steps on a Raspberry Pi 3B:
https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool

 >

woody-apple, 2021-02-03:
Are you able to take a peek at this @mspang is this a CIPD issue?

woody-apple, 2021-06-03:
I believe this is resolved, given things are working in our test events.

closed 2021-06-03 by woody-apple