project-chip/connectedhomeip #4468: Rename SecurePairingSession class and associated delegates

pan-apple, 2021-01-22:
#### Problem
SecurePairingSession is implements PASE protocol, as defined in CHIP specifications. The class name is not conveying the purpose of code.

 #### Proposed Solution
Rename SecurePairingSession class, and files to PasswordAuthenticatedSessionEstablishment
Reword pairing related functions in class and delegate (AuthenticatedSessionEstablishmentDelegate) to convey SessionEstablishment functionality.

turon, 2021-01-22:
Could we call them `PASESession` and `CASESession` for brevity?
Does the class just establish the session or does it manage the full lifecycle?

pan-apple, 2021-01-23:
Yes, sounds good to me. These classes are just establishing the session. The session is handed over to SecureSessionMgr.

PR #4491 (2021-01-26): Rename code to PASESession and CASESession
 #### Problem
SecurePairingSession implements PASE protocol, as defined in CHIP specifications. The class name is not conveying the purpose of code.

 #### Summary of Changes
Rename the code and files as follows
1. SecurePairingSession -> PASESession
2. CertificateAuthenticatedSessionEstablishment -> CASESession
3. AuthenticatedSessionEstablishmentDelegate -> SessionEstablishmentDelegate
4. Rename functions in AuthenticatedSessionEstablishmentDelegate class

 Fixes #4468
closed 2021-01-26 by rwalker-apple