project-chip/connectedhomeip #4493: Android Chip-Tool <-> Raspberrypi light app ble onCharacteristicWrite error

wissn84, 2021-01-26:
#### Problem
Android Chip-Tool(LG V30, Andorid 9) <-> Raspberrypi(Ubuntu Server 20.04) lligh tapp ble provisioning error

Version
Android - 9bcb03eab19c37471be12c09fa09d568c6864ff7
RP - 9bcb03eab19c37471be12c09fa09d568c6864ff7

The Log is below

01-26 14:58:05.907 D 1596     1690     chip.BluetoothManager:                                  CHIP-3840.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 1 
01-26 14:58:05.908 E 1596     1690     AndroidChipStack:                                       onCharacteristicWrite for 18ee2ef5-263d-4559-959f-4f9c429f9d11 failed with status: 1 

and Log is attacthed

I just change MTU Size 23 on android side

[LogFilter_20210126_145209.zip](https://github.com/project-chip/connectedhomeip/files/5871480/LogFilter_20210126_145209.zip)

yunhanw-google, 2021-03-08:
have you used bluez 5.55 on Ubuntu Server 20.10? the default bluez along with ubuntu server 20.04 does not work with chip

wissn84, 2021-03-11:
> have you used bluez 5.55 on Ubuntu Server 20.10? the default bluez along with ubuntu server 20.04 does not work with chip

Hi, I am using Ubuntu 20.10 and Default Bluez Version?
If upgrading Bluez, Does it work?
Thank you in advanced!

closed 2022-01-25 by woody-apple