project-chip/connectedhomeip #4504: Remove temporary replacements for NFC payload URI compatibility

kkasperczyk-no, 2021-01-26:
#### Problem
In the #4495 there were introduced some temporary replacements in NFC payload of spaces with _ for nrfconnect platform and _ with spaces for Android platform. It was done to not wait on fixing problems discussed in the CHIP spec issue #415.

 #### Proposed Solution
Remove replacements after fixing problems discussed in the CHIP spec issue #415.

turon, 2022-01-31:
Relevant portions of code related to this issue were tagged with TODO(#4504), but no reference to #4504 exists in current codebase.  This may mean this has been fixed, but we need to verify for v1 -- QR format must make consistent use of `_` or ` ` that matches spec.
Removing `v1_triage_split_4 `.

woody-apple, 2022-02-01:
Closing, given this will be verified by the CSG, we can re-open if this fails.

closed 2022-02-01 by woody-apple