project-chip/connectedhomeip #4505: Lighting app no longer compiles

bzbarsky-apple, 2021-01-26:
#### Problem
```
../../examples/lighting-app/linux/main.cpp:182:50: error: no member named 'StartWiFiManagement' in 'chip::DeviceLayer::ConnectivityManagerImpl'
        chip::DeviceLayer::ConnectivityMgrImpl().StartWiFiManagement();
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
../../examples/lighting-app/linux/main.cpp:187:28: error: no member named 'ThreadStackMgrImpl' in namespace 'chip::DeviceLayer'
        chip::DeviceLayer::ThreadStackMgrImpl().InitThreadStack();
        ~~~~~~~~~~~~~~~~~~~^
2 errors generated.
```
I get this if I just run `./gn_build.sh` at toplevel on Mac.  Note that this is compiling this app, but _not_ on Linux.

* `StartWiFiManagement` is only defined if `CHIP_DEVICE_CONFIG_ENABLE_WPA` but the use here is unconditional.   Note that `chip_enable_wifi` is only set to true in `src/platform/device.gni` if `chip_device_platform == "linux"`.
*  `examples/lighting-app/linux/main.cpp ` does not include `platform/ThreadStackManager.h`.  And even if it did there is no `platform/Darwin/ThreadStackManagerImpl.h`

 #### Proposed Solution
Fix the broken handling of the WPA ifdef.  Not sure what is supposed to happen with the thread stack manager.

@erjiaqing

vivien-apple, 2021-01-26:
Duplicate of #4472 I guess.

bzbarsky-apple, 2021-01-26:
I guess this is a duplicate of #4472

closed 2021-01-26 by bzbarsky-apple