project-chip/connectedhomeip #4519: Bootstrap on MacOS doesn't work anymore

pjzander-signify, 2021-01-27:
#### Problem
When running the bootstrap on MacOS the script is failing.
The error is introduced in commit 2244db (@szatmz )

 #### Error
  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done
  Setting up Python environment.....[|]
ERROR at //build/config/linux/pkg_config.gni:97:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /Users/paul/ConnectedHome/connectedhomeip/.environment/pigweed-venv/connectedhomeip-7c00447c524347032c311e74821f3908/
Command: python3 /Users/paul/ConnectedHome/connectedhomeip/build/config/linux/pkg-config.py openssl
Returned 1.
stderr:

Could not run pkg-config.

See //src/crypto/BUILD.gn:39:3: whence it was called.
  pkg_config("openssl_config") {
  ^-----------------------------
See //BUILD.gn:50:7: which caused the file to be included.
      "${chip_root}/src/crypto",
      ^------------------------

Traceback (most recent call last):
  File "/Users/paul/ConnectedHome/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 593, in <module>
    sys.exit(main())
  File "/Users/paul/ConnectedHome/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 585, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/paul/ConnectedHome/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 307, in setup
    result = step()
  File "/Users/paul/ConnectedHome/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 417, in virtualenv
    env=self._env,
  File "/Users/paul/ConnectedHome/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 225, in install
    install_packages(gn_target)
  File "/Users/paul/ConnectedHome/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 204, in install_packages
    ins.read())
subprocess.CalledProcessError: Command '('gn', 'gen', '/Users/paul/ConnectedHome/connectedhomeip/.environment/pigweed-venv/connectedhomeip-7c00447c524347032c311e74821f3908')' returned non-zero exit status 1
Error during bootstrap--see messages above.

wissn84, 2021-01-28:
@pjzander-signify 
You shall install openssl using brew then set pkg_config 

Please refer to BUILDING Document.
https://github.com/project-chip/connectedhomeip/blob/master/docs/BUILDING.md

`How to install prerequisites on macOS
On macOS, first install Xcode from the Mac App Store. The remaining dependencies can be installed and satisfied using Brew:

brew install openssl pkg-config
However, that does not expose the package to pkg-config. To fix that, one needs to run something like the following:

cd /usr/local/lib/pkgconfig
ln -s ../../Cellar/openssl@1.1/1.1.1g/lib/pkgconfig/* .
where openssl@1.1/1.1.1g may need to be replaced with the actual version of OpenSSL installed by Brew.

Note: If using MacPorts, port install openssl is sufficient to satisfy this dependency.

`

woody-apple, 2021-02-03:
I can confirm this seems to just work for me. @pjzander-signify can you confirm please - and reopen if still broken!

closed 2021-02-03 by woody-apple

xylophone21, 2021-11-08:
On M1, brew 3.3.2, seems there is no /usr/local/lib/pkgconfig any more?

brew -v
Homebrew 3.3.2
Homebrew/homebrew-core (git revision 19b41bdfac9; last commit 2021-11-08)

brew install openssl pkg-config
Updating Homebrew...
Warning: openssl@3 3.0.0_1 is already installed and up-to-date.
To reinstall 3.0.0_1, run:
  brew reinstall openssl@3
Warning: pkg-config 0.29.2_3 is already installed and up-to-date.
To reinstall 0.29.2_3, run:
  brew reinstall pkg-config

cd /usr/local/lib/pkgconfig
cd: no such file or directory: /usr/local/lib/pkgconfig

nekteckIOS, 2022-06-29:
> `brew info openssl` gave me the hint to set an environment variable which fixed it for me.
> 
> it told me:
> 
> ```
> For pkg-config to find openssl@3 you may need to set:
>   set -gx PKG_CONFIG_PATH "/opt/homebrew/opt/openssl@3/lib/pkgconfig"
> ```
> 
> since i am using fish this will probably be different for you

Are you finished with this issue? How is it solved?