project-chip/connectedhomeip #4527: EFR32 Build is blocking Openthread Update

jepenven-silabs, 2021-01-27:
#### Problem
When trying to update the OpenThread submodule, the EFR32 CI job fails, since the newest version of OpenThread use the new GSDK of silabs with new API.

 #### Proposed Solution
Update the GSDK that the EFR32 platform use for a newer version so we can update the OpenThread submodule.

tima-q, 2021-02-08:
@jepenven-silabs Any update on this update work ? #4561 can only progress by your update here.

jepenven-silabs, 2021-02-08:
> @jepenven-silabs Any update on this update work ? #4561 can only progress by your update here.

Yes, Should have two PR open this week to fix this. First one is to ensure the backward compatibility and second one is with the GSDK update.

jepenven-silabs, 2021-02-15:
@tima-q Quick update, need to fix a bug with BLE before I can open a PR.

PR #4870 (2021-02-17): Update EFR32 SDK from 2.7 to 3.1
 #### Problem
Cannot update the OpenThread submodule in Chip because EFR32 CI job fails. This is because of the API used in CHIP are older than the one used in OpenThread.

 #### Summary of Changes
Update EFR32 sdk used in CHIP to latest version for future OpenThread update
 Fixes #4527 
 Fixes #4278
 Fixes #2596
 Fixes #2595

For update : 
@tima-q @doru91
closed 2021-02-17 by mspang