project-chip/connectedhomeip #4531: Reporting:  If chip-tool does not initiate the connection to the server, incoming reports are dropped

woody-apple, 2021-01-27:
#### Problem
 If chip-tool does not initiate the connection to the server, incoming reports are dropped with: **CHIP: [IN] Data received on an unknown connection (0). Dropping it!!**

pan-apple, 2021-01-27:
The issue is caused by the unprovoked message (reports) from the server to chip-tool. This will be handled by CASE session setup mechanism. The flow of events will be

1. `chip-tool` perform the device pairing, and provisions operational credentials on the device.
2.  At some point in time, the device generates a report that will be sent to another node (e.g. `chip-tool`)
3. The device looks for an existing CASE session with the other node. If one exists, it'll use it. Otherwise, it'll setup a new CASE session using the provisioned operational credentials.
4. The CASE handshake will ensure that the other node (`chip-tool`) has a secure connection that can receive and decrypt the data.
5. The device will send the report using the secure session established via CASE.

Currently, the CASE handshake is not implemented in the code. I talked to @vivien-apple and he has a work around for the time being. So, maybe we can wait for CASE implementation to resolve this issue.

pan-apple, 2021-05-17:
The following PRs had some CASE related work done that's relevant to this issue.
https://github.com/project-chip/connectedhomeip/pull/6810
https://github.com/project-chip/connectedhomeip/pull/6791