project-chip/connectedhomeip #4544: Receiving first ZCL command from Android CHIP Tool always fails.

kkasperczyk-no, 2021-01-28:
#### Problem
When using Android CHIP Tool to control CHIP device running nrfconnect lock-app example there is a problem that sending first ON/OFF/TOGGLE command always results in data model processing failure. Following commands are processed properly and the first command is the only one that fails.

I noticed that source of a problem is Android CHIP Tool app and commit that seems to introduce it is:
https://github.com/project-chip/connectedhomeip/commit/fad6d0538fe1e6f906aa2e984f1fc3d02ecac001

When using Android CHIP Tool in previous revision, everything works fine. Revision of the lock-app firmware doesn't seem to have impact on it.

Below there are logs demonstrating problem. As you can see first message (the bolden one) is dropped due to invalid endpoint ID, which is different every time. Also RX len is bigger (25) than expected (3). There is also second message attached to show the proper message structure.
I: 22929[SVR] Packet received from UDP:2001:db8:2:0:88c5:d4a1:a7a6:5f77:46027: zu bytes
I: 22937[ZCL] APS frame processing success!                                                                                   
**I: 22941[ZCL] RX len 25, ep 13, clus 0x38a9                                                                                   
I: 22945[ZCL]  FC ea seq e8 cmd 9c payload[                                                                                   
I: 22949[ZCL] 0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX0hX                                              I: 22956[ZCL] ]                                                                                                                                                                                                                   
I: 22961[ZCL] Drop cluster 0x38a9 command 0x9c                                                                                
I: 22965[ZCL]  due to invalid endpoint:                                                                                       
I: 22969[ZCL] 0x13                                                                                                            
I: 22970[ZCL] Data model processing failure!**                                                                                  
I: 29304[SVR] Packet received from UDP:2001:db8:2:0:88c5:d4a1:a7a6:5f77:46027: zu bytes                                       
I: 29311[ZCL] APS frame processing success!                                                                                   
I: 29315[ZCL] RX len 3, ep 1, clus 0x 6                                                                                       
I: 29319[ZCL]  FC 1 seq 1 cmd 2 payload[                                                                                      
I: 29323[ZCL] NULL                                                                                                            
I: 29324[ZCL] ]                                                                                                                                                                                                                            I: 29329[ZCL] On/Off set value: 1 2                                                                                           
I: 29332[ZCL] Toggle on/off from 1 to 0                                                                                       
I: Unlock Action has been initiated  

 #### Proposed Solution

jrhees-cae, 2021-01-30:
I see this as well. Notable, the packet size of the first APS payload is large (25 bytes), when a typical On/Off/toggle APS payload is only 3 bytes.

LuDuda, 2021-01-31:
@pan-apple could you look at it?

kkasperczyk-no, 2021-02-01:
@LuDuda, as far as I know @pan-apple is already working on it, but he has some problems with reproducing it, as we use nRF and Android platforms. @jrhees-cae do you use the same platforms, as described or something else? It might help in finding problem source.

jrhees-cae, 2021-02-01:
I have only tested this recently on the Android platform, so I cannot comment on IOS.

PR #4643 (2021-02-04): Load secure session parameters before sending message to the device
 #### Problem
After pairing the device, the first message sent by the controller is corrupted when received by the device. This happen only in certain use cases, where the UDP transport information in not updated in the peer connection table.

 #### Summary of Changes
1. Check peer connection table entry for transport type. If the transport is incorrect, reload the security parameter for the connection.
2. The packet at re-send is corrupted since the original packet got modified (encryption, and header attachment) before the o [... cut]
closed 2021-02-04 by andy31415