project-chip/connectedhomeip #4547: Ble connection success callback multiple times on darwin platform

erjiaqing, 2021-01-28:
#### Problem

It is called every time it discovers a service on a peripheral. Which should be called once only when it connected.

https://github.com/project-chip/connectedhomeip/blob/b0fcbd2944c838245356ae2720a9463555959d57/src/platform/Darwin/BleConnectionDelegateImpl.mm#L192-L200

 #### Proposed Solution

Fix it.

erjiaqing, 2021-01-28:
Assigned to @vivien-apple from code history, please help redirect this to correct person if needed.

vivien-apple, 2021-03-05:
@erjiaqing Do you have more details ? For example I can't reproduce with the tuple `chip-tool/M5Stack`.

Looking at the code, the only obvious way it can happens is if the device advertise twice with the same short service `UUID` or if `PairDevice` is called twice.

tcarmelveilleux, 2022-01-31:
This is obsolete.

closed 2022-01-31 by tcarmelveilleux