project-chip/connectedhomeip #4558: CMake Error: build lock-app nrfconnect example

RoyAnupam, 2021-01-29:
#### Problem

Facing build error when trying to build onnectedhomeip/examples/lock-app/nrfconnect

 * **expected behavior** : Should build successfully
 * **actual behavior** : 
 Make Error at /home/r/ncs/zephyr/cmake/toolchain/gnuarmemb/generic.cmake:16 (message):
  Nothing found at GNUARMEMB_TOOLCHAIN_PATH: 'toolchain-dir'
 * steps to reproduce: Try to build using west for nRF52480DK board
 nrfconnect$ west build -b nrf52840dk_nrf52840

 * **system configuration**
Distributor ID:	Ubuntu
Description:	Ubuntu 20.10
Release:	20.10
Codename:	groovy

 * **screenshots, images, logs, etc.**
 
lock-app/nrfconnect$ west build -b nrf52840dk_nrf52840
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /home/r/ncs/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/r/work/CHIP/connectedhomeip/examples/lock-app/nrfconnect
-- Zephyr version: 2.4.0 (/home/r/ncs/zephyr)
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: /home/r/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
**CMake Error at /home/r/ncs/zephyr/cmake/toolchain/gnuarmemb/generic.cmake:16 (message):**
  **Nothing found at GNUARMEMB_TOOLCHAIN_PATH: 'toolchain-dir'**
Call Stack (most recent call first):
  /home/r/ncs/zephyr/cmake/generic_toolchain.cmake:65 (include)
  /home/r/ncs/zephyr/cmake/app/boilerplate.cmake:589 (include)
  /home/r/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/r/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:26 (find_package)

-- Configuring incomplete, errors occurred!
**FATAL ERROR: command exited with status 1: /**home/r/work/CHIP/connectedhomeip/.environment/cipd/pigweed/bin/cmake -DWEST_PYTHON=/home/r/work/CHIP/connectedhomeip/.environment/pigweed-venv/bin/python -B/home/r/work/CHIP/connectedhomeip/examples/lock-app/nrfconnect/build -S/home/r/work/CHIP/connectedhomeip/examples/lock-app/nrfconnect -GNinja

***Additional Information:**
Ran following commands to set environment variables before building the application

- nrfconnect$ source ~/ncs/zephyr/zephyr-env.s
- rfconnect$ export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
- nrfconnect$ export GNUARMEMB_TOOLCHAIN_PATH="~/gnuarmemb"

~/ncs : It is nRF SDK instllation directory
~/gnuarmemb: It is my GNU Arm Embedded Toolchain directory

**Note:** I have  downloaded Tolchain : gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux.tar.bz2
from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads URL

Any suggestion about how to fix the above issue will be really helpful. Thank You

BR,
-Anupam Roy

msandstedt, 2021-01-29:
You'll need to install the cross-compile toolchain.

In OSX:

```
# nRF Toolchain Installation (OSX)

# install brew
/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"

# install arm cross-compile tools
brew tap PX4/homebrew-px4
brew update
brew install gcc-arm-none-eabi

# install dependencies
brew install cmake ninja gperf python3 ccache qemu dtc

# install nrf command-line tools and JLink
cd ~/
curl https://www.nordicsemi.com/-/media/Software-and-other-downloads/Desktop-software/nRF-command-line-tools/sw/Versions-10-x-x/10-12-0/nRF-Command-Line-Tools_10_12_0_OSX.tar -o nRF-Command-Line-Tools_10_12_0_OSX.tar
tar -xf nRF-Command-Line-Tools_10_12_0_OSX.tar
sudo installer -verbose -pkg nRF-Command-Line-Tools_10_12_0_OSX.pkg -target /
sudo installer -verbose -pkg JLink_MacOSX_V688a.pkg -target /
cd -
```

In Ubuntu:

```
# nRF Toolchain Installation (Ubuntu)

# install dependencies
sudo apt-get install libncurses5 gcc-arm-none-eabi gcc-arm-none-eabi
# Note: for manual install, gcc-arm-none-eabi also available here: https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/10-2020q4/gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux.tar.bz2

# go home
cd ~/

# install gn
git clone https://gn.googlesource.com/gn
cd gn
python build/gen.py
ninja -C out
echo "export PATH=${PATH}:${PWD}/out" >> ~/.bashrc
source ~/.bashrc
cd -

# install nrf command-line tools and JLink
curl https://www.nordicsemi.com/-/media/Software-and-other-downloads/Desktop-software/nRF-command-line-tools/sw/Versions-10-x-x/10-12-1/nRFCommandLineTools10121Linuxamd64tar.gz -o nRFCommandLineTools10121Linuxamd64tar.gz
tar -xf nRFCommandLineTools10121Linuxamd64tar.gz
sudo dpkg -i nRF-Command-Line-Tools_10_12_1_Linux-amd64.deb
sudo dpkg -i JLink_Linux_V688a_x86_64.deb
```

Then export:

```
# export arm tools variant and location
ARM_NONE_EABI_GCC=$(which arm-none-eabi-gcc)
ARM_NONE_EABI_GCC_DIR=$(dirname $ARM_NONE_EABI_GCC)
ARM_NONE_EABI_GCC_LINK=$(readlink $ARM_NONE_EABI_GCC)
export GNUARMEMB_TOOLCHAIN_PATH=$(dirname ${ARM_NONE_EABI_GCC_DIR}/${ARM_NONE_EABI_GCC_LINK})/../
export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
```

woody-apple, 2021-02-03:
Marking as closed, given the above. Please re-open if you're still hitting issues!

closed 2021-02-03 by woody-apple

RoyAnupam, 2021-02-03:
@msandstedt , Thank You very much for your kind help! The issue is resolved by following your guide.