project-chip/connectedhomeip #4581: Can't compile nrfconnect lighting app

bzbarsky-apple, 2021-01-30:
#### Problem
When I follow the steps in `examples/lighting-app/nrfconnect/README.md` to get nrfconnect 1.4.0, update deps (like zephyr), and then `west build -b nrf52840dk_nrf52840` I get build errors like so inside zephyr due to warnings that become fatal with #4528:
```
tools/nrfconnect/zephyr/subsys/shell/shell_fprintf.c: In function 'shell_fprintf_buffer_flush':
tools/nrfconnect/zephyr/subsys/shell/shell_fprintf.c:47:53: error: pointer targets in passing argument 2 of 'sh_fprintf->fwrite' differ in signedness [-Werror=pointer-sign]
   47 |  sh_fprintf->fwrite(sh_fprintf->user_ctx, sh_fprintf->buffer,
      |                                           ~~~~~~~~~~^~~~~~~~
      |                                                     |
      |                                                     uint8_t * const {aka unsigned char * const}
tools/nrfconnect/zephyr/subsys/shell/shell_fprintf.c:47:53: note: expected 'const char *' but argument is of type 'uint8_t * const' {aka 'unsigned char * const'}
....
tools/nrfconnect/zephyr/subsys/shell/shell_help.c: In function 'help_item_print':
tools/nrfconnect/zephyr/subsys/shell/shell_help.c:113:37: error: pointer targets in passing argument 1 of 'strlen' differ in signedness [-Werror=pointer-sign]
  113 |  const uint16_t offset = 2 * strlen(tabulator) + item_name_width + 1;
      |                                     ^~~~~~~~~
      |                                     |
      |                                     const uint8_t * {aka const unsigned char *}
In file included from tools/nrfconnect/zephyr/include/sys/ring_buffer.h:16,
                 from tools/nrfconnect/zephyr/include/shell/shell_history.h:13,
                 from tools/nrfconnect/zephyr/include/shell/shell.h:12,
                 from tools/nrfconnect/zephyr/subsys/shell/shell_ops.h:10,
                 from tools/nrfconnect/zephyr/subsys/shell/shell_help.c:7:
tools/gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi/include/string.h:41:17: note: expected 'const char *' but argument is of type 'const uint8_t *' {aka 'const unsigned char *'}
   41 | size_t  strlen (const char *);
      |                 ^~~~~~~~~~~~
cc1: all warnings being treated as errors
```
etc, etc.  I assume this compiles in our CI somehow, but I'm not sure what the difference is...

@Damian-Nordic 

 #### Proposed Solution
Either turn off -Werror for these files or figure out how to get them to be warning-free.

bzbarsky-apple, 2021-02-01:
@kkasperczyk-no do you have any idea why I might be hitting this while CI does not?

kkasperczyk-no, 2021-02-01:
@bzbarsky-apple are you using macOS for compiling? I tried to build it on Linux and CI also runs build on Linux, where it works fine. At first glance it looks like some gcc version issue, but I'm using exactly the same, as you and I don't see warnings. I don't have clear explanation why it could make a difference between macOS and Linux, but if you use mac, I will try to get one tomorrow and try to find a reason.

bzbarsky-apple, 2021-02-01:
@kkasperczyk-no Yes, I am using a Mac.

LuDuda, 2021-02-03:
I will revert `-Werror` till we update Zephyr base. Thanks for reporting.

PR #4648 (2021-02-04): [nrfconnect] revert adding -Werror flag
#### Problem
Zephyr does not compile without warnings on macOS.

#### Summary of Changes
Revert part of #4528 till warnings are fixed with new update.

Fixes #4581
closed 2021-02-04 by woody-apple