project-chip/connectedhomeip #4635: TestExchangeManager::CheckExchangeMessages crashes when malicious packet is sent

pan-apple, 2021-02-03:
#### Problem
There seems to be some issue in `TestExchangeManager::CheckExchangeMessages`. It's working on tip of master at commit fcc0e36d26110b23cacedbd702d4e9dcc024e783. But, it is because the acknowledgement for the malicious packet is being sent to the wrong NodeID. See the logs below.

```
CHIP: [IN] Secure message was encrypted: Msg ID 0
CHIP: [IN] Sending msg from 123654 to 111222333
CHIP: [IN] Sending secure msg on generic transport
CHIP: [EM] ec pos: 3, id: 16809, Delegate: 0x0
CHIP: [EM] Dropping unexpected message 00000001:2 41A9 MsgId:00000000
CHIP: [EM] Piggybacking Ack for MsgId:00000000 with msg
CHIP: [IN] Secure message was encrypted: Msg ID 0
CHIP: [IN] Sending msg from 123654 to 123654
CHIP: [IN] Sending secure msg on generic transport
CHIP: [IN] Data received on an unknown connection (2). Dropping it!!
CHIP: [EM] Accept FAILED, err = Error 4127 (0x0000101F)
CHIP: [IN] Secure msg send status 0
CHIP: [EM] Flushed pending ack for MsgId:00000000
CHIP: [IN] Secure msg send status 0
CHIP: [IN] Secure message was encrypted: Msg ID 1
CHIP: [IN] Sending msg from 123654 to 111222333
CHIP: [IN] Sending secure msg on generic transport
CHIP: [EM] ec pos: 3, id: 16809, Delegate: 0xe9d9c508
CHIP: [IN] Secure msg send status 0
````

Specifically, `CHIP: [IN] Sending msg from 123654 to 123654`, where the CRMP acknowledgement is sent from a node to itself. The message is dropped as the receiver does not identify the connection corresponding to the key ID.

If the connection tracking logic is fixed, the test starts crashing. The following are the test logs (which shows the src and dest node is correct for the ack message).
```
CHIP: [IN] Secure message was encrypted: Msg ID 0
CHIP: [IN] Sending msg from 123654 to 111222333
CHIP: [IN] Sending secure msg on generic transport
CHIP: [EM] ec pos: 3, id: 16809, Delegate: 0x0
CHIP: [EM] Dropping unexpected message 00000001:2 41A9 MsgId:00000000
CHIP: [EM] Piggybacking Ack for MsgId:00000000 with msg
CHIP: [IN] Secure message was encrypted: Msg ID 0
CHIP: [IN] Sending msg from 111222333 to 123654
CHIP: [IN] Sending secure msg on generic transport
CHIP: [EM] Rxd Ack; Removing MsgId:00000000 from Retrans Table
CHIP: [EM] Removed CHIP MsgId:00000000 from RetransTable
CHIP: [IN] Secure msg send status 0
CHIP: [EM] Flushed pending ack for MsgId:00000000
CHIP: [IN] Secure msg send status 0
Abort trap: 6
```

and, the following is the stack trace for the crash.
```
  * frame #0: 0x00007fff67cd733a libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff67d93e60 libsystem_pthread.dylib`pthread_kill + 430
    frame #2: 0x00007fff67c5e808 libsystem_c.dylib`abort + 120
    frame #3: 0x000000010000f605 TestExchangeMgr`chip::Messaging::ReliableMessageMgr::StartRetransmision(this=0x0000000100059040, entry=0x0000000100059070) at ReliableMessageMgr.cpp:333:5
    frame #4: 0x000000010000be10 TestExchangeMgr`chip::Messaging::ExchangeContext::SendMessage(this=0x000000010005b6c0, protocolId=1, msgType='\x02', msgBuf=PacketBufferHandle @ 0x00007ffeefbff520, sendFlags=0x00007ffeefbff518, msgCtxt=0x0000000000000000) at ExchangeContext.cpp:160:52
```
 #### Proposed Solution
Debug and fix the issue.

yufengwangca, 2021-04-15:
@woody-apple @pan-apple 

Where is TestExchangeManager::CheckExchangeMessages, is this the source of test harness tools? I want to check how they use ExchangeMgr API

yufengwangca, 2021-04-16:
From the log, it show the crash occurs within ReliableMessageMgr::StartRetransmision(), possibly either the entry or entry->rc is null. This indicate the exchange might already be freed during CRMP re-transmission. 

I suspect the test code use Release or Free to free the exchange after it complete sending message, but I am not sure. It is quite difficult for me to root cause of this issue further without source code. More detailed information is needed, such as where to check the source code of test tool and how to setup the environment to repo this issue

pan-apple, 2021-04-16:
So, if you enable the commented out code in the CheckExchangeMessages test, it’ll start crashing.
```
--- a/src/messaging/tests/TestExchangeMgr.cpp
+++ b/src/messaging/tests/TestExchangeMgr.cpp
@@ -150,8 +150,9 @@ void CheckExchangeMessages(nlTestSuite * inSuite, void * inContext)
     // send a malicious packet
     // TODO: https://github.com/project-chip/connectedhomeip/issues/4635
-    // ec1->SendMessage(0x0001, 0x0002, System::PacketBufferHandle::New(System::PacketBuffer::kMaxSize),
-    //                 SendFlags(Messaging::SendMessageFlags::kNone));
+    ec1->SendMessage(Protocols::Id(VendorId::Common, 0x0001), 0x0002,
+                     System::PacketBufferHandle::New(System::PacketBuffer::kMaxSize),
+                     SendFlags(Messaging::SendMessageFlags::kNone));
     NL_TEST_ASSERT(inSuite, !mockUnsolicitedAppDelegate.IsOnMessageReceivedCalled);
     // send a good packet
```

Currently, the crash stack is different than what’s in the issue. The underlying code has changed (the issue is from 1st week of Feb).

yufengwangca, 2021-04-19:
@pan-apple 

Thanks for sharing the repo procedure, the root cause of this crash is when the malicious packet is looped back from the the LoopbackTransport, ExchangeMgr treats it as a valid message sent from the a know exchange, and deliver to the exchange delegate which actual send the message.

Summary of Changes
The exchangeMgr should only deliver the received message to the exchange delegate when the received message's source Node ID matches the peer Node ID associated with the exchange, or the peer Node ID of the exchange is 'any'

bzbarsky-apple, 2021-04-28:
Fix was reverted.

closed 2021-05-07 by yufengwangca