project-chip/connectedhomeip #4654: Need Example in CI for Mbed OS platform

woody-apple, 2021-02-04:
#### Problem
Our build examples in GitHub don't include the Mbed OS. We should have one of these for all platforms to make sure we build.

 #### Proposed Solution
Make one!

woody-apple, 2021-02-04:
@lukgni Can you take a peek at this as well please? This will get your platform integrated into our CI. There are plenty of examples up there, and if you need help - please reach out and we can help you set this up.

lukgni, 2021-02-04:
@woody-apple Right! I will discuss this item with our team. 
As far as I know, we want to port  shell, lighting and (maybe) lock example to Mbed OS. These set should cover most of a CHIP functionalities and could be a nice proof-of-concept for us.
Personally I think shell example will be that one which we could submit the fastest.

closed 2022-10-08 by stale[bot]