project-chip/connectedhomeip #4660: Prerequisites for building iOS project

virupaksha, 2021-02-04:
Dear team,

I have checkout CHIP repo into my mac machine and would like to run iOS project But unfortunately project fails to build(Error : "Command PhaseScriptExecution failed with a nonzero exit code"). As I don't see any document to build the project or may be I am not able to find it. Could you please someone help me to solve this issue.

woody-apple, 2021-02-04:
Hi there! Have you tried out the documentation here? It's required to build even from Xcode!

https://github.com/project-chip/connectedhomeip/blob/master/docs/BUILDING.md

Thanks

woody-apple, 2021-02-15:
Closing, given this seems to work. Please re-open if you have any issues!

closed 2021-02-15 by woody-apple

virupaksha, 2021-03-09:
Hi there,

Below is the output after I hit the command "source scripts/bootstrap.sh" on terminal. I see the below error is blocking me to 
proceed further. May I what can be the issue.
/*******************
Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done
  Setting up Python environment.....Traceback (most recent call last):
  File "/Users/virupakshfutane/Desktop/CHIP/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 598, in <module>
    sys.exit(main())
  File "/Users/virupakshfutane/Desktop/CHIP/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 590, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/virupakshfutane/Desktop/CHIP/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 308, in setup
    result = step()
  File "/Users/virupakshfutane/Desktop/CHIP/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 416, in virtualenv
    env=self._env,
  File "/Users/virupakshfutane/Desktop/CHIP/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 256, in install
    install_packages(gn_target)
  File "/Users/virupakshfutane/Desktop/CHIP/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 225, in install_packages
    stderr=outs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 185, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 5] Input/output error
Error during bootstrap--see messages above.
virupakshfutane@MXDEAPSEP8HTD6L CHIP % 
***********************************/

virupaksha, 2021-03-09:
Thank you for the support I am able to figure out the issue and it's been fixed :). Thank you for the support

saurabhst, 2021-03-09:
> Thank you for the support I am able to figure out the issue and it's been fixed :). Thank you for the support

@virupaksha would you mind mentioning how did you resolve the issue just in case if some one else bumps into similar issue in future ? That would be really helpful.

virupaksha, 2021-03-10:
@saurabhst I had issue with "cylancePROTECT" agent which is running on my Macbook pro and it was blocking me. I just disabled it.