project-chip/connectedhomeip #4664: Python crash in nl_Chip_DeviceController_DeleteDeviceController

andy31415, 2021-02-04:
#### Problem
 
 Python crashes on exit if trying to connect to a BLE device and failing.
 I was never able to really connect to a raspberry pi. When attempting on my linux machine, I fail to connect but exiting with ^D I get a `SEGV`:
 
 ```
 Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007ffff61bc030 in chip::app::Command::Shutdown (this=0x6233613638336166) at ../../src/app/Command.cpp:130
130	    VerifyOrExit(mState != kState_Uninitialized, );
(gdb) bt
#0  0x00007ffff61bc030 in chip::app::Command::Shutdown() (this=0x6233613638336166) at ../../src/app/Command.cpp:130
#1  0x00007ffff61b75c1 in chip::Controller::Device::~Device() (this=0xb47588, __in_chrg=<optimized out>) at ../../src/controller/CHIPDevice.h:64
#2  0x00007ffff61b7ebe in chip::Controller::DeviceController::~DeviceController() (this=0xb47288, __in_chrg=<optimized out>) at ../../src/controller/CHIPDeviceController.h:112
#3  0x00007ffff61b7fbb in chip::Controller::DeviceCommissioner::~DeviceCommissioner() (this=0xb47288, __in_chrg=<optimized out>) at ../../src/controller/CHIPDeviceController.h:250
#4  0x00007ffff61bb0a2 in chip::DeviceController::ChipDeviceController::~ChipDeviceController() (this=0xb47270, __in_chrg=<optimized out>) at ../../src/controller/CHIPDeviceController_deprecated.cpp:82
#5  0x00007ffff61bb0ca in chip::DeviceController::ChipDeviceController::~ChipDeviceController() (this=0xb47270, __in_chrg=<optimized out>) at ../../src/controller/CHIPDeviceController_deprecated.cpp:82
#6  0x00007ffff6176f30 in nl_Chip_DeviceController_DeleteDeviceController(chip::DeviceController::ChipDeviceController*) (devCtrl=0xb47270) at ../../src/controller/python/ChipDeviceController-ScriptBinding.cpp:141
#7  0x00007ffff7fbfd1d in  () at /lib/x86_64-linux-gnu/libffi.so.7
#8  0x00007ffff7fbf289 in  () at /lib/x86_64-linux-gnu/libffi.so.7
#9  0x00007ffff760256a in _ctypes_callproc () at /usr/lib/python3.8/lib-dynload/_ctypes.cpython-38-x86_64-linux-gnu.so
#10 0x00007ffff7601e2c in  () at /usr/lib/python3.8/lib-dynload/_ctypes.cpython-38-x86_64-linux-gnu.so
#11 0x0000000000510603 in _PyObject_MakeTpCall ()
#12 0x000000000050abf2 in _PyEval_EvalFrameDefault ()
#13 0x000000000051ae40 in _PyFunction_Vectorcall ()
#14 0x00000000005636cd in  ()
#15 0x00000000006076d7 in  ()
#16 0x00000000004f5506 in  ()
#17 0x00000000005f4b2b in _PyGC_CollectNoFail ()
#18 0x00000000005f3e90 in PyImport_Cleanup ()
#19 0x00000000005efebb in Py_FinalizeEx ()
#20 0x00000000005fea98 in Py_Exit ()
#21 0x00000000005f585b in  ()
#22 0x00000000005f4e52 in PyErr_PrintEx ()
#23 0x00000000005f3444 in PyRun_SimpleFileExFlags ()
#24 0x00000000005e8775 in Py_RunMain ()
#25 0x00000000005c6f19 in Py_BytesMain ()
#26 0x00007ffff7c46d0a in __libc_start_main (main=0x5c6ee0 <main>, argc=2, argv=0x7fffffffe2d8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe2c8) at ../csu/libc-start.c:308
#27 0x00000000005c6e1a in _start ()
```

andy31415, 2021-02-04:
I am unclear what sets `mCommandSender` to null to start with.

We should force to initialize all members of classes.

erjiaqing, 2021-02-05:
The mCommandSender is initialized as nullptr

https://github.com/project-chip/connectedhomeip/blob/65f94f90d72ac5d2584ba0facb2fa401d15a3f7f/src/controller/CHIPDevice.h#L297

erjiaqing, 2021-02-05:
`this=0x6233613638336166`

>>> b'\x62\x33\x61\x36\x38\x33\x61\x66'
b'b3a683af'

I guess this is caued by buffer overrun.

andy31415, 2021-02-05:
> The mCommandSender is initialized as nullptr
> 
> https://github.com/project-chip/connectedhomeip/blob/65f94f90d72ac5d2584ba0facb2fa401d15a3f7f/src/controller/CHIPDevice.h#L297

That was a patch I added - I was not the case when I got the crash and I agree that it is very likely the cause of the issue.

andy31415, 2021-02-05:
Marking as closed for now on the assumption that mCommandSender setting to nullptr fixed it. Will reopen if it happens again.

closed 2021-02-05 by andy31415