project-chip/connectedhomeip #4672: Builds action failure

pan-, 2021-02-04:
#### Problem

`Builds` workflow seems to be broken at the moment: https://github.com/project-chip/connectedhomeip/actions?query=workflow%3ABuilds . 
It displays the same error in forks. Comparing logs, some things have changed: 
- Ubuntu: `20210123.1` -> `20210131.1`
- Docker: `1.40` -> `1.41`
- CodeQL: `2.4.0` -> `2.4.2`

It seems that `TestInetLayerDNS` is also now run. It was not the case in my previous log.
What is surprising is that unchanged code now fails.  

 #### Proposed Solution

Identify the root cause of the issue and fix it.

woody-apple, 2021-02-04:
https://github.com/project-chip/connectedhomeip/issues/4670

woody-apple, 2021-02-04:
(Working on it now!)

closed 2021-02-04 by woody-apple