project-chip/connectedhomeip #4694: [chip-tool] [M5Stack] Basic on/off test not working for IPv6

turon, 2021-02-05:
#### Problem
Setup an initial quick start setup with chip-tool and all-clusters-app on M5Stack.
On/off commands work over IPv4 but not IPv6.

Pass:
```
./out/debug/standalone/chip-tool pairing bypass 192.168.86.59 11097
./out/debug/standalone/chip-tool onoff on 1
./out/debug/standalone/chip-tool onoff off 1

CHIP: [ZCL] DefaultResponse:
CHIP: [ZCL]   ClusterId: 0x0006
CHIP: [ZCL]   CommandId: 0x01
CHIP: [ZCL]   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
CHIP: [TOO] Default Success Response
CHIP: [ZCL] Data model processing success!
```

Fail:
```
./out/debug/standalone/chip-tool pairing bypass fe80:0000:0000:0000:9af4:abff:fe6b:1814 11097
./out/debug/standalone/chip-tool onoff on 1
./out/debug/standalone/chip-tool onoff off 1

CHIP: [TOO] No response from device
CHIP: [CTL] Shutting down the commissioner
CHIP: [CTL] Shutting down the controller
CHIP: [TOO] Run command failure: Error 4172 (0x0000104C)

```

turon, 2021-02-05:
Tested with `chip-tool` on Mac.

wissn84, 2021-02-09:
I think To use IPv6 You shall include interface behind ipv6 address.
for example fe80:0000:0000:0000:9af4:abff:fe6b:1814%<interface> -> fe80:0000:0000:0000:9af4:abff:fe6b:1814%eth0 
You can check using ifconfig command.

sweetymhaiske, 2021-04-08:
@turon Can you please try it with the latest commit ID bc3d6bf593ed27592a939a58a056776b6a5407bb. I checked it on my side it worked fine for me.

franck-apple, 2021-04-20:
not reproducible. Closing.

closed 2021-04-20 by franck-apple