project-chip/connectedhomeip #4701: On-mesh prefix not persistent on OTBR

Ramin2094, 2021-02-05:
#### Problem
On-mesh prefix isn't persistent on Open Thread Border Router after reboot. I already found an issue from openthread page ( https://github.com/openthread/wpantund/issues/342 ) for this topic. The response for this issue was that the application layer needs to care of it.

 #### Proposed Solution
An additional tool needs to be run on Raspberry Pi (OTBR) to make such data persistent.

jwhui, 2021-02-05:
What we really need here is for the Thread Border Router to automatically configure the network to enable IPv6 reachability between Thread and Wi-Fi/Ethernet links. Having to manually configure an on-mesh prefix is not a workable solution for the consumer market. This issue applies more broadly than just CHIP.

There is existing work to enable this:
- https://github.com/openthread/openthread/pull/5856
- https://github.com/openthread/openthread/pull/6008

woody-apple, 2021-02-15:
Closing given comments above, sounds like these are OT issues?

closed 2021-02-15 by woody-apple