project-chip/connectedhomeip #4736: raspi: Linux app fails to pair with python controller

cjandhyala, 2021-02-08:
Linux app (Lighting)  fails to pair with python controller, fails at PASE SPAKE2 handshake 
========== Controller side Log ============
chip-device-ctrl > connect -ble 3840 12345678
CHIP: [BLE] NewConnection
Rendezvous Initialized
Connected
chip-device-ctrl > CHIP: [BLE] CBManagerState: ON
CHIP: [BLE] Connecting to device with discriminator: 3840
CHIP: [BLE] Warning: OnBleConnectionComplete is called multiple times for the same peripheral.
CHIP: [BLE] subscribe complete, ep = 0x103c2c7c8
CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
CHIP: [BLE] local and remote recv window size = 3
CHIP: [BLE] Sent PBKDF param request
CHIP: [BLE] Received pairing message from 12344321
CHIP: [BLE] Received pairing message for 112233
CHIP: [BLE] Received PBKDF param response
CHIP: [BLE] Sent spake2p msg1
CHIP: [BLE] Received pairing message from 12344321
CHIP: [BLE] Received pairing message for 112233
CHIP: [BLE] Received spake2p msg2
CHIP: [BLE] Sent spake2p msg3
CHIP: [CTL] Remote device failed in SPAKE2+ handshake

CHIP: [BLE] Auto-closing end point's BLE connection.

========== Device side log ============
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffb0024920
CHIP:BLE: Indication confirmation, 0xffffb0024920
CHIP:BLE: Indication confirmation, 0xffffb0024920
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffb0024920
CHIP:BLE: Indication confirmation, 0xffffb0024920
CHIP:BLE: Indication confirmation, 0xffffb0024920
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffb0024920
CHIP:BLE: Received rendezvous message from 112233
CHIP:BLE: Received spake2p msg3
CHIP:SVR: OnRendezvousStatusUpdate: Error 4172 (0x0000104C)
CHIP:SVR: OnRendezvousError: Error 4172 (0x0000104C)
CHIP:SVR: OnRendezvousStatusUpdate: Error 4172 (0x0000104C)
CHIP:SVR: OnRendezvousError: Error 4172 (0x0000104C)
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffb0024920
CHIP:SVR: OnRendezvousError: Error 4003 (0x00000FA3)
CHIP:BLE: Indication confirmation, 0xffffb0024920
CHIP:BLE: Indication confirmation, 0xffffb0024920
CHIP:BLE: ack recv timeout, closing ep 0xaaaad2579d78
CHIP:SVR: OnRendezvousError: Ble Error 6024 (0x00001788): Message fragment acknowledgement timed out
CHIP:BLE: Releasing end point's BLE connection back to application.
CHIP:BLE: Got notification regarding chip connection closure

erjiaqing, 2021-02-09:
```
CHIP:BLE: Received spake2p msg3
CHIP:SVR: OnRendezvousStatusUpdate: Error 4172 (0x0000104C)
```

@pan-apple , Hi Pankaj, seems the Linux app failed to finish key exchange, do you have any ideas about how to debug this? Previously, I met this error and it is resolved by removing `/tmp/chip*`, and I believe that is caused by `https://github.com/project-chip/connectedhomeip/issues/4713`.

However, as for this issue, @cjandhyala told me that the /tmp/chip* files are empty, , can you reproduce this error on your side?

pan-apple, 2021-02-09:
Hi Song, I don't have a Linux setup to reproduce it on my end. Let me check with @cjandhyala and see if we can get more debug information to narrow down the issue.

cjandhyala, 2021-02-10:
Issue is not reproduced after updating the code to the latest.

closed 2021-02-10 by cjandhyala