project-chip/connectedhomeip #4836: Sort out saner naming/placement for our "safe cast" helpers

bzbarsky-apple, 2021-02-12:
#### Problem
We have `src/lib/core/CHIPSafeCasts.h` (which does safe uint8_t <-> char casting) and `src/lib/support/SafePointerCast.h` which allows safe (for some values of safe) casting between two structs.

 #### Proposed Solution
We should probably at least have our "safe cast" files living in the same directory.  We should also consider either giving `CHIPSafeCasts` a less generic name or combining the files.

@Damian-Nordic

Damian-Nordic, 2021-02-12:
@bzbarsky-apple Ah, I didn't notice we had `CHIPSafeCasts.h`. I can fix it, but do you or anyone else understand the difference between `support` and `core` directories? I believe utility classes suits better the `support` directories, which don't depend on anything CHIP-specific, but `core` also contains things like `Optional` and I'm confused.

bzbarsky-apple, 2021-02-12:
@Damian-Nordic I don't know what the intended difference is....  And to be clear, I have no strong opinion about finer-grained headers vs a single "safe casts" header; just in the former case we should give the uint-to-char one a better name.

@andy31415 Do you?  Would be good to toss some `README.md` in things to document what they're for.

gerickson, 2021-02-12:
These (`support` vs. `core`) are latent artifacts of the original Weave tree layout that seeded the Project CHIP layout. `support` was code that was used across all of the various `*Layers` whereas `core` was intended to be the code that supported the Weave _core_ messaging and protocol layer but not `profile` initiator and responder implementation.

Consequently, I'd agree that if there is anything in `core` in CHIP that's used more broadly, it should be factored out of `core` and either put in `support` or some other "shared" directory.