project-chip/connectedhomeip #4837: Ensure that we properly process (e.g. by dropping early enough in the process) messages we don't understand

bzbarsky-apple, 2021-02-12:
#### Problem
The CHIP spec just got clarified to say that various currently-reserved values in the message header should lead to the message being dropped, because we don't know what to do with it.  See https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/1814

 #### Proposed Solution
Ensure that our message processing aligns with the spec.

362dyx, 2021-03-12:
I met the same problem and didn't understand the reason.I can't open this link（ CHIP-Specifications/connectedhomeip-spec#1814）. Can you give me a new one

bzbarsky-apple, 2021-03-13:
@362dyx Unfortunately the spec is not public at the moment.  You have to be a member of the Zigbee Alliance to access it.  See https://github.com/project-chip/connectedhomeip/pull/5019#issuecomment-795782952 for details.

turon, 2022-01-31:
Reserved Version != 0 and DSIZ = b11 values are handled in `MessageHeader.cpp`:

```
   VerifyOrExit(version == kMsgHeaderVersion, err = CHIP_ERROR_VERSION_MISMATCH);

    // ...
   
    if (mMsgFlags.HasAll(Header::MsgFlagValues::kDestinationNodeIdPresent, Header::MsgFlagValues::kDestinationGroupIdPresent))
    {
        // Reserved.
        SuccessOrExit(err = CHIP_ERROR_INTERNAL);
    }
```

Security Flags are deprecated.

Closing.  Removed tag `v1_triage_split_4`.

closed 2022-01-31 by turon