project-chip/connectedhomeip #4861: TCPBase::ProcessReceivedBuffer seems to loop forever if the message can't fit in the first buffer in the chain

bzbarsky-apple, 2021-02-15:
#### Problem
`TCPBase::ProcessReceivedBuffer` is handed a possibly-chained `PacketBuffer`.  It then does the following (somewhat abridged):

1. Read the message length (which is the first two bytes of a message, always).
2. Check whether the data in the first buffer in the chain is at least that much.
3. If not, and we have a chain, compact head and try again.

What happens if the first buffer in the chain is not large enough for the whole message?  It seems like this could easily happen if a smaller-than-the-message packet comes in on the wire and we allocate a buffer sized for it.  Then more data comes in, and we chain more things onto that buffer, but we can't squeeze the whole message into that first buffer, so the above code seems like it would lead to an infinite loop...

Am I missing something here?  Or is this actually a problem we need to address?

 #### Proposed Solution
Either document why the "try again" loop is guaranteed to terminate or fix the code so it in fact terminates.

@kpschoedel @andy31415 @pan-apple

kpschoedel, 2021-02-16:
I think the original assumption was that the receive packet buffer was maximum-size and therefore a complete message would always fit. So for valid messages, I broke this assumption adding `RightSize()` to `TCPEndPoint::ReceiveData()` and maybe that should be reverted. But also, there's no test for excessive size (corrupt/malicious size field) and that will always fail `ContainsCompleteMessage()`.

bzbarsky-apple, 2021-02-16:
I'm not sure anything in the spec imposes a size cap on messages over a TCP transport.  So even a spec-compliant implementation seems like it could send something longer than our packetbuffer maxsize.

PR #5265 (2021-03-19): Handle large messages in TCPBase::ProcessReceivedBuffer
#### Problem

`TCPBase::ProcessReceivedBuffer()` seems to loop forever if the message
can't fit in the first buffer in the chain.

#### Summary of Changes

Functionality:
* Pass only a single linear buffer, containing exactly the message, up the stack (rather than a shared reference to `TCPEndPoint.mRcvBuf` that could cause trouble if higher levels were to do things like altering the chain). 
* If a message can not be put in a linear buffer (either because the system is out of memory or because it's too large) return a error; the data is  [... cut]
commit 4eb50c2b (2021-03-19): Handle large messages in TCPBase::ProcessReceivedBuffer (#5265)

* Handle large messages in TCPBase::ProcessReceivedBuffer

#### Problem

TCPBase::ProcessReceivedBuffer seems to loop forever if the message
can't fit in the first buffer in the chain.

#### Summary of Changes

If a message does not fit in the current buffer but would fit in a
larger buffer, allocate a larger buffer.

If a message would not fit in a maximum-sized buffer, discard it.

Fixes #4861 - TCPBase::ProcessReceivedBuffer seems to loop forever if the message can't fit in the first buffer in the chain

* Add unit test and fi [... cut]
closed 2021-03-19 by andy31415