project-chip/connectedhomeip #4917: Chip device controller running inside docker container unable to see bluetooth adapter.

SaileeG, 2021-02-18:
Setup: Docker container running on Raspi 4 Host

Unable to see bluetooth adapter in chip device controller running inside docker container. 
root@ubuntu:/connectedhomeip/examples/lighting-app/linux# chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233
 
Chip Device Controller Shell
 
chip-device-ctrl > ble-adapter-print
chip-device-ctrl > 
chip-device-ctrl >

Expected output(As seen on Raspi Host):
ubuntu@ubuntu:~/connectedhomeip/examples/chip-tool/out/debug$ chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

Chip Device Controller Shell

chip-device-ctrl > ble-adapter-print
2021-02-18 06:54:52,379 ChipBLEMgr   INFO     adapter 0 = B8:27:EB:C4:E2:93

The bluetooth adapter is seen inside the docker container: 
root@ubuntu:/connectedhomeip/examples/lighting-app/linux# hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:C4:E2:93  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN 
	RX bytes:5670 acl:0 sco:0 events:336 errors:0
	TX bytes:6940 acl:0 sco:0 commands:228 errors:0
 
root@ubuntu:/connectedhomeip/examples/lighting-app/linux# hcitool dev
Devices:
	hci0	B8:27:EB:C4:E2:93
root@ubuntu:/connectedhomeip/examples/lighting-app/linux# ls -l /sys/class/bluetooth
total 0
lrwxrwxrwx 1 root root 0 Feb 17 02:21 hci0 -> ../../devices/platform/soc/fe201000.serial/tty/ttyAMA0/hci0

mspang, 2021-02-23:
Don't run it in docker.

SaileeG, 2021-02-25:
This is as a part of our simulated devices section of CHIP certification tool
The goal was to be able to run a docker container for every non DUT entity needed for certification testing

As a part of which we are trying to validate communication between:

- Docker container and the DUT(nordic dev boards)
- Docker container and the host

Being able to run lighting or lock sample app inside a docker container to achieve simulated light or lock endpoint etc. which can communicate with the Python controller running on the host for scenarios like BLE pairing etc.

SaileeG, 2021-02-26:
Created Virtual BLE interfaces using Bluez emulator. Able to see 2 Virtual interfaces(hci1 and hci2) in the docker container1, also able to see only 1 virtual BLE interface hci2 when running ble-adapter-print in chip device controller running inside the docker container 1. 

Facing an issue when trying to use virtual interfaces inside 2 docker containers on the same host:

- In both docker containers I see 2 virtual interfaces in hciconfig.
- In Container1:
    - I am able to see only hci2 when running ble-adapter-print from chip-device-ctrl
    - I am able to run Linux Lighting app using BLE interface hci2 only
- In Container2:
    - I am unable to see any ble interfaces when running ble-adapter-print from chip-device-ctrl
    - I am unable to run Linux Lighting app using any BLE interface

For more details check: https://docs.google.com/document/d/17dRNe5oM5xbKMYZ3tZqioWTI61qwr-KQkCeSq4M1gUI/edit?resourcekey=0-LiWqT3cnRrzrHTHDHl324A#heading=h.vje4z92f9fiq

mspang, 2021-02-26:
This seems like quite a different problem than the original description describes (what's described there seems expected to me, not a bug).

mspang, 2021-02-26:
It seems unlikely to me that providing the same HCI device (virtual or otherwise) to more than one container would work correctly.

mspang, 2021-02-26:
@yunhanw-google has experience with bluez and may be able to help

SaileeG, 2021-02-26:
@mspang thats right we are no longer seeing the original issue after creating the virtual BLE interfaces using Bluez. I will create 2 Separate issues for the single and multi container issue described in the above comment

SaileeG, 2021-03-01:
Single Container Issue: https://github.com/project-chip/connectedhomeip/issues/5045
Multi Container issue: https://github.com/project-chip/connectedhomeip/issues/5047

SaileeG, 2021-03-23:
#5047 and #5045 resolved

closed 2021-03-23 by SaileeG