project-chip/connectedhomeip #4918: BluezPeripheralRegister fail due to BLE adapter issue seen when running Linux Lighting sample app example inside Docker container running on Raspi

SaileeG, 2021-02-18:
Setup: Docker container running on Raspi 4 Host

Lighting app Linux example:
We need to run the following before using chip-tool for making the dbus and bt adaptor available to the tool:
```
service dbus start
bluetoothd &
```
 
```
root@ubuntu:/connectedhomeip/examples/lighting-app/linux# out/debug/chip-tool-server --ble-device 0
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*700 0C9SS0]
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 12344321
 
CHIP:DL: wpa_supplicant: _IsWiFiStationProvisioned: interface not connected
CHIP:BLE: Waiting for PBKDF param request
CHIP:DIS: CHIP minimal mDNS configured as 'Commisioning device'.
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bus acquired for name CHIP-3840
CHIP:DL: CREATE service object at /chipoble/002a/service
CHIP:DL: Create characteristic object at /chipoble/002a/service/c1
CHIP:DL: Create characteristic object at /chipoble/002a/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/002a/service
CHIP:DL: CHIP BTP C2 /chipoble/002a/service
CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is TRUE
CHIP:DL: Create characteristic object at /chipoble/002a/service/c3
CHIP:DL: CHIP BTP C3 /chipoble/002a/service
CHIP:DL: FAIL: NULL apEndpoint->mpAdapter in bluezObjectsSetup
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: FAIL: NULL endpoint->mpAdapter in BluezPeripheralRegisterApp
CHIP:DIS: MinMdns received a query.
CHIP:DIS: QUERY IN/PTR: 192.168.0.176.in-addr.arpa.
CHIP:DIS: MinMdns received a query.
CHIP:DIS: QUERY IN/PTR: 192.168.0.176.in-addr.arpa.
CHIP:DIS: MinMdns received a query.
```

Current issue: 
CHIP:DL: FAIL: NULL endpoint->mpAdapter in BluezPeripheralRegisterApp

Expected output:
```
ubuntu@ubuntu:~/connectedhomeip/examples/lighting-app/linux$ sudo out/debug/chip-tool-server --ble-device 0
CHIP:DL: writing settings to file (/tmp/chip_factory.ini.tmp)
CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
CHIP:DL: NVS set: chip-factory/pin-code = 12345678 (0xBC614E)
CHIP:DL: writing settings to file (/tmp/chip_factory.ini.tmp)
CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
CHIP:DL: NVS set: chip-factory/discriminator = 1383 (0x567)
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*K2H5+B9SS0]
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 12344321
 
CHIP:BLE: Waiting for PBKDF param request
CHIP:DIS: CHIP minimal mDNS configured as 'Operational device'.
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:IN: New pairing for device 112233, key 0!!
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bus acquired for name CHIP-1383
CHIP:DL: CREATE service object at /chipoble/e0a7/service
CHIP:DL: Create characteristic object at /chipoble/e0a7/service/c1
CHIP:DL: Create characteristic object at /chipoble/e0a7/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/e0a7/service
CHIP:DL: CHIP BTP C2 /chipoble/e0a7/service
CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is TRUE
CHIP:DL: Create characteristic object at /chipoble/e0a7/service/c3
CHIP:DL: CHIP BTP C3 /chipoble/e0a7/service
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent , 16387
CHIP:DL: Create adv object at /chipoble/e0a7/advertising
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x00, 0x67, 0x05, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent , 16388
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DIS: MinMdns received a query.
CHIP:DIS: QUERY IN/PTR: 192.168.0.100.in-addr.arpa.
CHIP:DIS: MinMdns received a query.
```

Likely related to: https://github.com/project-chip/connectedhomeip/issues/4917

mspang, 2021-02-23:
Don't run it in docker.

closed 2021-02-23 by mspang

SaileeG, 2021-02-25:
This is as a part of our simulated devices section of CHIP certification tool
The goal was to be able to run a docker container for every non DUT entity needed for certification testing

As a part of which we are trying to validate communication between:
- Docker container and the DUT(nordic dev boards)
- Docker container and the host

Goal is Being able to run lighting or lock sample app inside a docker container to achieve simulated light or lock endpoint etc. which can communicate with the Python controller running on the host for scenarios like BLE pairing etc.

SaileeG, 2021-02-26:
Lighting App running inside Container with virtual BLE interfaces: Works only with hci2(BLE virtual interface 2)
`root@ubuntu:/connectedhomeip/examples/lighting-app/linux# out/debug/chip-tool-server --ble-device 2
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*700 0C9SS0]
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 12344321
 
CHIP:DL: wpa_supplicant: _IsWiFiStationProvisioned: interface not connected
CHIP:BLE: Waiting for PBKDF param request
CHIP:DIS: CHIP minimal mDNS configured as 'Commisioning device'.
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bus acquired for name CHIP-3840
CHIP:DL: CREATE service object at /chipoble/2f67/service
CHIP:DL: Create characteristic object at /chipoble/2f67/service/c1
CHIP:DL: Create characteristic object at /chipoble/2f67/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/2f67/service
CHIP:DL: CHIP BTP C2 /chipoble/2f67/service
CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is TRUE
CHIP:DL: Create characteristic object at /chipoble/2f67/service/c3
CHIP:DL: CHIP BTP C3 /chipoble/2f67/service
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent 16391
CHIP:DL: Create adv object at /chipoble/2f67/advertising
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent 16392
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: RegisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent 16393
CHIP:DIS: MinMdns received a query.
CHIP:DIS: QUERY IN/PTR: _AAF8F49E._sub._googlecast._tcp.local.
CHIP:DIS: QUERY IN/PTR: _googlecast._tcp.local.
CHIP:DIS: MinMdns received a query.
CHIP:DIS: QUERY IN/PTR: _AAF8F49E._sub._googlecast._tcp.local.
CHIP:DIS: QUERY IN/PTR: _googlecast._tcp.local.
CHIP:DIS: MinMdns received a query.
`
CHIP device controller run on Raspi host to scan the Linux Light app running inside the Container
`chip-device-ctrl > ble-scan
2021-02-25 15:34:10,479 ChipBLEMgr   INFO     scanning started
2021-02-25 15:34:14,728 ChipBLEMgr   INFO     Name            = CHIP-3840
2021-02-25 15:34:14,728 ChipBLEMgr   INFO     ID              = 2193682d-4caf-3a4a-9713-03a1a3329db8
2021-02-25 15:34:14,730 ChipBLEMgr   INFO     RSSI            = 127
2021-02-25 15:34:14,731 ChipBLEMgr   INFO     Address         = 00:AA:01:01:00:24
2021-02-25 15:34:14,736 ChipBLEMgr   INFO     Pairing State   = 0
2021-02-25 15:34:14,736 ChipBLEMgr   INFO     Discriminator   = 3840
2021-02-25 15:34:14,736 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-02-25 15:34:14,736 ChipBLEMgr   INFO     Product Id      = 65279
2021-02-25 15:34:14,739 ChipBLEMgr   INFO     Adv UUID        = 0000feaf-0000-1000-8000-00805f9b34fb
2021-02-25 15:34:14,741 ChipBLEMgr   INFO     Adv Data        = 00000f5a23fffe
2021-02-25 15:34:14,741 ChipBLEMgr   INFO     
2021-02-25 15:34:24,831 ChipBLEMgr   INFO     scanning stopped
chip-device-ctrl > 
chip-device-ctrl > ble-connect CHIP-3840
2021-02-25 15:36:15,198 ChipBLEMgr   INFO     trying to connect to CHIP-3840
2021-02-25 15:36:17,227 ChipBLEMgr   INFO     BLE connecting
2021-02-25 15:36:17,228 ChipBLEMgr   INFO     Discovering services
2021-02-25 15:36:17,347 ChipBLEMgr   INFO     Service discovering success
2021-02-25 15:36:17,369 ChipBLEMgr   INFO     connect success
`
More details and Linux lighting app logs can be found at: https://docs.google.com/document/d/17dRNe5oM5xbKMYZ3tZqioWTI61qwr-KQkCeSq4M1gUI/edit?usp=sharing&resourcekey=0-LiWqT3cnRrzrHTHDHl324A

The issues I am facing now are:
 1) In first Container The HCI2(BLE Virtual interface 2) is the only one which is available/can be used to run the Linux Lighting app
 2) In a Second container I am able to see both the virtual interfaces in hciconfig but unablee to run lighting app using any of the ble interfaces